OllisGit/OctoPrint-FilamentManager #1: Try to rebuild Filament Database

digitaljunk, 2020-09-13:
I lost an SD card due power loss and needed to rebuild my octopi installation. I have some backups taken the other day (oprint, .octoprint Directories) but I have no simple export of the FilamentManager Filament database. I try to find the database in my backups. Can someone please point me to it? Or am I lost?

NotExpectedYet, 2020-09-19:
> I lost an SD card due power loss and needed to rebuild my octopi installation. I have some backups taken the other day (oprint, .octoprint Directories) but I have no simple export of the [...]

It's in your .octoprint folder somewhere I believe. Do you have a plugin data folder? I forget what it looks like.

OllisGit, 2020-09-22:
Hi @digitaljunk,
look for this file:  /OctoPrint/data/filamentmanager/filament.db.
BR
Olli

digitaljunk, 2020-09-22:
> Hi @digitaljunk,
> look for this file: /OctoPrint/data/filamentmanager/filament.db.
> BR
> Olli
Hi @OllisGit,
Just perfect. Saved me a lot of time. Thank you! 
BR Rob

closed 2020-09-22 by OllisGit