OllisGit/OctoPrint-FilamentManager #2: Print Cost is missing

fractalsauce, 2020-09-18:
So first of all this is my first time ever submitting a bug on GitHub so forgive me if my post seems lacking.

**Describe the bug**
Print cost should be visible in the left-hand sidebar under "State", but it is not. I have filled out the filament profiles with all of the required fields including cost per spool and weight.

**Steps to reproduce**
Install filament manager, configure filament profiles, upload a gcode file, select gcode.

All other relevant information is visible, but the "Cost" entry doesn't even show, let alone the price.

**Expected behavior**
View the print cost for the selected print under "State" in the sidebar.

**Did the same happen when all other 3rd party plugins are disabled?**
Yes

**Log file**

[octoprint.log](https://github.com/OllisGit/OctoPrint-FilamentManager/files/5248123/octoprint.log)

**Screenshots**

[Plugin Manager](https://i.imgur.com/yP1LtPt.png)

[State Sidebar](https://i.imgur.com/sl0eikQ.png)

**Additional context**
I'm not really sure what it would be. I originally thought it was a conflict with one of my other plugins. I didn't even know you could disable individual plugins without removing them until I started filling out this bug report. I thought that would fix it and help me narrow down the issue, but now I have no idea.

Any help would be appreciated, thanks!

NotExpectedYet, 2020-09-19:
Do you have [](https://plugins.octoprint.org/plugins/costestimation) installed? 

I believe that is required for the cost to show.

fractalsauce, 2020-09-19:
Wow I feel stupid! I didn't know it needed that and I don't see it mentioned anywhere on the [FilamentManager](https://github.com/malnvenshorn/OctoPrint-FilamentManager) main page. Maybe that should be updated for new users of the plugin such as myself.

Unfortunately I just started an 11 hour print so I'm unable to install CostEstimation at the moment to test it out. But as soon as it finishes I'll give it a shot and update this issue.

Thanks for the help!

fractalsauce, 2020-09-20:
Finally got it working! For anyone with this issue in the future, since the original CostEstimation plugin has been abandoned, download and install @benjamin-salchow's fork via the plugin manager using the following link:

https://github.com/benjamin-salchow/OctoPrint-CostEstimation/archive/master.zip

closed 2020-09-20 by fractalsauce