OllisGit/OctoPrint-FilamentManager #4: External Database not working

Moskito99, 2020-09-26:
**Describe the bug**
Can't connect to a external Postgresql Database

**Steps to reproduce**
1. Fill out external information
2. Click on Connection test and nothing happend....

**Expected behavior**
Connection

**Did the same happen when all other 3rd party plugins are disabled?**
Don't know

**Log file**
`2020-09-26 17:26:21,885 - tornado.access - WARNING - 400 POST /plugin/filamentmanager/database/test (::ffff:192.168.2.104) 30.44ms`

**Screenshots**
[image]

**Additional context**
OctoPrint 1.4.2, OctoPi 0.17.0, Filament Manager 1.6.0

jeremyhaynes217, 2020-09-26:
I've got the same problem and the same error and I've been trying to figure it out for 4 hours now and I think Google is out of suggestions.

jeremyhaynes217, 2020-09-27:
So I found [https://community.octoprint.org/t/filament-manager-plugin-how-to-setup-postgresql-database/10698](https://community.octoprint.org/t/filament-manager-plugin-how-to-setup-postgresql-database/10698) and followed the whole thing to the letter and it still didn't work. I finally gave up and turned off my printer (and consequently my raspberry pi because I have them all attached to an Alexa-controlled smart plug. I came back 5 hours later and turned it back on to see if I could export from postgres to mysql or just use the filament.db from sqlite in Grafana.... long story short, without doing anything since I gave up other than turn it on,  I clicked "Test connection" because I'm a masochist like that, and it turned green, and everything worked. 
[image]
[image]

Moskito99, 2020-09-27:
OK, after a few hours I found the problem!

The documentation in the readme file is not clear enough!

I've installed the psycopg2 via OS and not via `~/oprint/bin/pip install psycopg2`!
So I got this message:
`2020-09-27 08:24:16,027 - octoprint.plugins.filamentmanager - ERROR - Failed to initialize database: No module named 'psycopg2'`

The log entry was strange and so I read your hint with the this link:
[https://community.octoprint.org/t/filament-manager-plugin-how-to-setup-postgresql-database/10698](https://community.octoprint.org/t/filament-manager-plugin-how-to-setup-postgresql-database/10698)

After the installation of psycopg2 via `~/oprint/bin/pip install psycopg2`, the external database is working now!
Still interesting, that is working without the port of the database...
[image]

After all, it is working now for me....

closed 2020-09-27 by Moskito99