OllisGit/OctoPrint-FilamentManager #5: Export not working

Moskito99, 2020-09-26:
**Describe the bug**
Database Export not working.

**Steps to reproduce**
1. Click on Export
2. Nothing happend expect eexport.html

**Expected behavior**
A database export?!?!

**Did the same happen when all other 3rd party plugins are disabled?**
Don't know.

**Log file**
`2020-09-26 17:34:22,602 - tornado.application - ERROR - Uncaught exception GET /plugin/filamentmanager/export?apikey= (::ffff:192.168.2.104)
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/tornado/web.py", line 1569, in _execute
    result = self.prepare()
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/server/util/tornado.py", line 284, in prepare
    self._fallback(self.request, b"")
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/server/util/tornado.py", line 648, in __call__
    response.extend(app_response)
  File "/home/pi/oprint/lib/python3.7/site-packages/werkzeug/wsgi.py", line 506, in __next__
    return self._next()
  File "/home/pi/oprint/lib/python3.7/site-packages/werkzeug/wrappers/base_response.py", line 45, in _iter_encoded
    for item in iterable:
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_filamentmanager/api/__init__.py", line 327, in file_generator
    for c in f:
  File "/home/pi/oprint/lib/python3.7/codecs.py", line 322, in decode
    (result, consumed) = self._buffer_decode(data, self.errors, final)
UnicodeDecodeError: 'utf-8' codec can't decode byte 0x8c in position 11: invalid start byte
2020-09-26 17:34:22,606 - tornado.access - ERROR - 500 GET /plugin/filamentmanager/export?apikey= (::ffff:192.168.2.104) 43.94ms
`

**Additional context**
OctoPrint 1.4.2, OctoPi 0.17.0, Filament Manager 1.6.0, Python 3.7

OllisGit, 2020-10-07:
Hi @Moskito99 ,

issue is fixed in latest version 1.6.1. Please retest and give feedback.

Thx, in advance
Olli

Moskito99, 2020-10-07:
Thank You!
It is working now!

closed 2020-10-07 by Moskito99