OllisGit/OctoPrint-FilamentManager #6: Hide spools

poutnik, 2020-09-29:
Hello, thanks for this great plugin.
I'd like to request a feature to allow me to hide a spool from the list on the sidebar, yet not to delete the spool entirely from the database - I have some filaments that I really like, I buy them repeatedly, but I sometimes don't have them - would be great if I could hide them, yet keep the (best) settings I already have found...

Thanks for consideration

Jiri

jeremyhaynes217, 2020-11-01:
I came here to ask for basically the same thing. I'd like an option to turn off any spools that are at 0 remaining. I need to keep them in the database so my Grafana dashboard can use it to calculate the cost of old prints, but I don't need to see them in the drop down in the sidebar or in the pop up.

elpeter25, 2020-11-29:
I arrived here looking for the same feature. It would be great to be able to hide spools from drop-down list either manually selecting them in settings or automatically when they reach 0.

Thanks and keep up the good work!

elpeter25, 2020-12-30:
Hi,
Still no news for this suggestion? It would we awesome to be able to hide empty spools from the list.

Cheers!

Ajimaru, 2020-12-30:
Would be a handy feature.

OllisGit, 2021-01-12:
Hi,

this feature is planed for the SpoolManager-Plugin, see https://github.com/OllisGit/OctoPrint-SpoolManager/issues/81

PS. Sorry, for being late to the party, but I made a "Octoprint sabatical" (no coding, no printing) the last months.

closed 2021-01-28 by OllisGit