OllisGit/OctoPrint-FilamentManager #7: Filament Manager can no longer access postgresql database on external Raspberry after Python 3 Upgrade

mpd352, 2020-09-30:
**Describe the bug**
A clear and concise description of what the bug is.

After the upgrade of my two Octoprint instances I can no longer use the shared postgresql database. Neither instance of Octoprint can connect to the external database. The log shows the statement "tornado.access - WARNING - 400 POST /plugin/filamentmanager/database/test (::ffff:192.168.0.195) 35.89ms"

My research indicates that this is some kind of confusion with the API key. I am not however accomplish enough user to figure out exactly how to get around this.

**Steps to reproduce**
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Did the same happen when all other 3rd party plugins are disabled?**
If not which plugin is causing the issue?

**Log file**
Drag and drop the octoprint.log here.
2020-09-30 02:54:31,448 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2020-09-30 02:54:31,453 - octoprint.server - INFO - --- Log roll over detected ---------------------------------------------------
2020-09-30 02:54:31,453 - octoprint.server - INFO - OctoPrint 1.4.2
2020-09-30 02:54:31,460 - octoprint.plugin.core - INFO - 33 plugin(s) registered with the system:
|  Action Command Notification Support (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/action_command_notification
|  Action Command Prompt Support (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/action_command_prompt
|  Announcement Plugin (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/announcements
|  Anonymous Usage Tracking (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/tracking
|  Application Keys Plugin (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/appkeys
|  Backup & Restore (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/backup
|  Consolidate Temp Control (0.1.7) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_consolidate_temp_control
|  Core Wizard (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/corewizard
|  Creality Temperature (1.2.4) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_CrealityTemperature
|  Cura Thumbnails (0.2.0) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_ultimakerformatpackage
|  Discovery (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/discovery
|  DisplayLayerProgress Plugin (1.23.3) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_DisplayLayerProgress
|  Error Tracking (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/errortracking
|  Filament Manager (1.6.0) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_filamentmanager
|  File Check (2020.8.7) (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_file_check
|  Firmware Check (2020.9.23) (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_firmware_check
|  Fullscreen Plugin (0.0.5) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_fullscreen
|  GCode Viewer (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/gcodeviewer
|  Logging (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/logging
|  Login UI (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/loginui
|  MultiCam (0.2.8) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_multicam
|  OctoPrint-WideScreen (0.1.3) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_widescreen
|  Pi Support Plugin (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/pi_support
|  Plugin Manager (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/pluginmanager
|  Preheat Button (0.5.1) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_preheat
|  PrettyGCode (1.2.2) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_prettygcode
|  PrintJobHistory (1.9.0) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_PrintJobHistory
|  Procastinator (1.1.1) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_procastinator
|  PrusaSlicer Thumbnails (0.1.2) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_prusaslicerthumbnails
|  Smart Preheat (0.0.5) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_smartpreheat
|  Software Update (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/softwareupdate
|  Virtual Printer (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/virtual_printer
| *Webcam Tab (0.1.2) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_webcamtab
Prefix legend: ! = disabled, # = blacklisted, * = incompatible
2020-09-30 02:54:31,479 - octoprint.environment - INFO - Detected environment is Python 3.7.3 under Linux (linux). Details:
|  hardware:
|      cores: 4
|      freq: 1200.0
|      ram: 916992000
|  os:
|      id: linux
|      platform: linux
|  plugins:
|      pi_support:
|          model: Raspberry Pi 3 Model B Rev 1.2
[... 355 lines cut ...]
2020-09-30 22:09:44,285 - octoprint.server.util.sockjs - INFO - New connection from client: ::ffff:192.168.0.114
2020-09-30 22:09:44,314 - octoprint.server.util.flask - INFO - Passively logging in user mpd352 from ::ffff:192.168.0.114
2020-09-30 22:09:44,314 - octoprint.access.users - INFO - Logged in user: mpd352
2020-09-30 22:09:46,049 - octoprint.plugins.announcements - INFO - Loaded channel _important from https://octoprint.org/feeds/important.xml in 0.6s
2020-09-30 22:09:46,302 - octoprint.plugins.PrintJobHistory - INFO - Plugin-State: PreHeat=enabled DisplayLayerProgress=enabled filamentmanager=enabled ultimakerformat=enabled PrusaSlicerThumbnails=enabled
2020-09-30 22:09:46,842 - octoprint.plugins.announcements - INFO - Loaded channel _releases from https://octoprint.org/feeds/releases.xml in 0.6s
2020-09-30 22:09:47,547 - octoprint.plugins.announcements - INFO - Loaded channel _blog from https://octoprint.org/feeds/octoblog.xml in 0.62s
2020-09-30 22:09:48,389 - octoprint.plugins.announcements - INFO - Loaded channel _plugins from https://plugins.octoprint.org/feed.xml in 0.79s
2020-09-30 22:09:49,047 - octoprint.plugins.announcements - INFO - Loaded channel _octopi from https://octoprint.org/feeds/octopi.xml in 0.6s
2020-09-30 22:09:49,933 - octoprint.plugins.pluginmanager - INFO - Loaded plugin notices data from https://plugins.octoprint.org/notices.json
2020-09-30 22:09:50,668 - octoprint.server.util.sockjs - INFO - User mpd352 logged in on the socket from client ::ffff:192.168.0.114
2020-09-30 22:24:31,557 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2020-09-30 22:24:39,475 - octoprint.plugins.tracking - INFO - Sent tracking event ping, payload: {'octoprint_uptime': 588609}
2020-09-30 22:30:43,425 - octoprint.server.util.sockjs - INFO - Client connection closed: ::ffff:192.168.0.114
2020-09-30 22:39:31,559 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2020-09-30 22:39:39,364 - octoprint.plugins.tracking - INFO - Sent tracking event ping, payload: {'octoprint_uptime': 589509}
2020-09-30 22:45:25,474 - tornado.access - WARNING - 400 POST /plugin/filamentmanager/database/test (::ffff:192.168.0.195) 35.89ms
2020-09-30 22:45:29,220 - tornado.access - WARNING - 400 POST /plugin/filamentmanager/database/test (::ffff:192.168.0.195) 32.55ms

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]

**Additional context**
Add any other context about the problem here.

mpd352, 2020-10-05:
Olli:

I spun up a clean instant of Octoprint and installed postgresql on it after updating everything on the instance except updating to Python3.  It's running Python2.7.  I'm running that as a standalone Octprint instance not attached to a printer, hosting the external filament database only.

I'm running two Octoprint instances, one attached to a Qidi X-One2 and a second attached to an Ender 3 Pro.  The Qidi octoprint has not been updated to Python3.  It immediately attached to the new external database.  

The Ender 3 Pro octoprint was upgraded to Phyton3, and was the original external database server.  It would not attached to the new external database server, and the non Python3 octoprint could not attach to it.

I spun up another new instant of octoprint for the Ender 3 Pro.  I updated everything on it except updating to Phython3.  It's running Python2.7.  It immediately connected to the new external database server.  

To summarize what I figured out is a client running Phython3 will not connect to an external database server period.  If the database server is running Python3, no clients can connect to it regardless of what version of Python they are running.

For a connection to an external database server to work neither the client or server can be running Python3.

The issue may be in the configuration of Filament Manager under Python3, but if so, I'm not smart enough to figure out what the issue is.

Thanks,

Kevin

OllisGit, 2020-10-07:
Hi @mpd352 ,

I could not reproduce the issue. I create a fresh database and a fresh OP with FilamentManager on Py3.6 and everything works as expected.

On which network port is your database running? Do you use the default port: 5432? If not, just add your port behind the IP-Adresss, like this: ```postgresql://192.168.0.73:6432```

If this is not the issue, please increase the log-level for FilamentManager and restart OP.
[image]

PS. next time, instead of pasting the full log-message to the textbox, just attach it. Thx, in advance

mpd352, 2020-10-07:
OllisGit:

My postgresql instances are running on port 5432, and I've tried it both with and without.  No joy.

OctoPI01-10-07-2020.log is from the server running my Qidi X-One2.  OctoPI02-10-07-2020.log is from the server running my Ender 3.  OctoPI03-10-07-2020.log is from the stand alone database server, it is not attached to any printer.

[OctoPI03-10-07-2020.log](https://github.com/OllisGit/OctoPrint-FilamentManager/files/5342992/OctoPI03-10-07-2020.log)
[OctoPI02-10-07-2020.log](https://github.com/OllisGit/OctoPrint-FilamentManager/files/5342999/OctoPI02-10-07-2020.log)
[OctoPI01-10-07-2020.log](https://github.com/OllisGit/OctoPrint-FilamentManager/files/5343014/OctoPI01-10-07-2020.log)

Thanks for the help

OllisGit, 2020-10-07:
Hi @mpd352 ,

I looked into your provided log files and found the main reasons why the plugin is not working:
```
OctoPI01-10-07-2020.log
=======================
2020-10-07 11:14:23,730 - octoprint.plugins.filamentmanager - ERROR - Failed to initialize database: (psycopg2.OperationalError) could not connect to server: Network is unreachable
	Is the server running on host "192.168.0.75" and accepting
	TCP/IP connections on port 5432?

OctoPI02-10-07-2020.log
=======================
2020-10-07 11:15:34,920 - octoprint.plugins.filamentmanager - ERROR - Failed to initialize database: No module named 'psycopg2'

OctoPI03-10-07-2020.log
=======================
2020-10-07 11:13:25,693 - octoprint.plugins.filamentmanager - ERROR - Failed to initialize database: No module named 'psycopg2'
```
01) network issue: connect to the raspi and try to connect to the database via telnet like this: ```telnet ip port```
02, 03) database module is not installed in your python environment. In most cases the module ```psycopg2``` is installed during plugin installation, but sometimes OP has some hickups and you need to install it manually.
try this:
```
~/OctoPrint/venv/bin/pip install psycopg2
```
More details, see https://github.com/OllisGit/OctoPrint-FilamentManager/issues/4 or in the wiki
https://github.com/OllisGit/OctoPrint-FilamentManager/wiki/Setup-PostgreSQL-on-Raspbian-(Stretch)

mpd352, 2020-10-07:
OllisGit

~/OctoPrint/venv/bin/pip install psycopg2 returned an error of command not found. A little bit of poking around Google led me to believe that there was no virtual environment installed.

I filched these lines from another article on the OctoPrint forum:

cd ~
sudo apt update
sudo apt install python-pip python-dev python-setuptools python-virtualenv git libyaml-dev build-essential
cd OctoPrint
virtualenv venv
source venv/bin/activate
~/OctoPrint/venv/bin/pip install psycopg2

And that seemed to install psycopg2

Still doesn't work. I've attached new logs.

Thank you,
Kevin Conde
Peoria, AZ

OllisGit, 2020-10-08:
Hi @mpd352,
you answered with the mail-client instead of using the GitHub-UI -> there is no attachment.

mpd352, 2020-10-08:
Hi OllisGit:

Sorry, here they are.  Thanks for you help.

[OctoPI02-10-07-2020.log](https://github.com/OllisGit/OctoPrint-FilamentManager/files/5348994/OctoPI02-10-07-2020.log)
[OctoPI03-10-07-2020.log](https://github.com/OllisGit/OctoPrint-FilamentManager/files/5348995/OctoPI03-10-07-2020.log)

OllisGit, 2020-10-08:
Same issue (No module named 'psycopg2'), because OP use a different python environment.

In the logs you can find this entry
```virtualenv: /home/pi/oprint```
So, you need to install the package in this environment.
```
    cd /home/pi/oprint/bin
    source activate

Show all modules
    pip list

Search for psycopg2
    pip list | grep  psycopg2

Install
    pip install psycopg2-binary 
```
See also the forum entry:
https://community.octoprint.org/t/filament-manager-cant-connect-to-postgresql-db-after-python-3-update/24773/10

mpd352, 2020-10-08:
OllisGit:

That worked!  I did have to make one minor change to the last command from:

pip install psycopg2-binary

to:

sudo pip install psycopg2-binary 

to get around a failed permission error.

Thank you,

Kevin.

closed 2020-10-08 by OllisGit

scaytrase, 2020-10-08:
Actually for similar reason I cannot upgrade the plugin to the latest release

```
2020-10-09 02:06:38,024   /home/pi/oprint/bin/python2 -m pip --disable-pip-version-check install https://github.com/OllisGit/OctoPrint-FilamentManager/releases/latest/download/master.zip --no-cache-dir
2020-10-09 02:06:40,061 > Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
2020-10-09 02:06:40,063 > Collecting https://github.com/OllisGit/OctoPrint-FilamentManager/releases/latest/download/master.zip
2020-10-09 02:06:41,065 > Downloading https://github.com/OllisGit/OctoPrint-FilamentManager/releases/latest/download/master.zip (245kB)
2020-10-09 02:06:42,447 > Requirement already satisfied: OctoPrint in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint-FilamentManager==1.6.2) (1.4.2)
2020-10-09 02:06:42,448 > Requirement already satisfied: backports.csv<1.1,>=1.0.5 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint-FilamentManager==1.6.2) (1.0.7)
2020-10-09 02:06:42,449 > Requirement already satisfied: uritools<2.2,>=2.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint-FilamentManager==1.6.2) (2.1.1)
2020-10-09 02:06:42,449 > Requirement already satisfied: SQLAlchemy<1.2,>=1.1.15 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint-FilamentManager==1.6.2) (1.1.18)
2020-10-09 02:06:44,179 > Collecting psycopg2-binary
2020-10-09 02:06:44,180 > Downloading https://files.pythonhosted.org/packages/fc/51/0f2c6aec5c59e5640f507b59567f63b9d73a9317898810b4db311da32dfc/psycopg2-binary-2.8.6.tar.gz (384kB)
2020-10-09 02:06:45,184 ! ERROR: Command errored out with exit status 1:
2020-10-09 02:06:45,185 ! command: /home/pi/oprint/bin/python2 -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-5CmQNJ/psycopg2-binary/setup.py'"'"'; __file__='"'"'/tmp/pip-install-5CmQNJ/psycopg2-binary/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' egg_info --egg-base /tmp/pip-install-5CmQNJ/psycopg2-binary/pip-egg-info
2020-10-09 02:06:45,186 ! cwd: /tmp/pip-install-5CmQNJ/psycopg2-binary/
2020-10-09 02:06:45,186 ! Complete output (23 lines):
2020-10-09 02:06:45,186 ! running egg_info
2020-10-09 02:06:45,187 ! creating /tmp/pip-install-5CmQNJ/psycopg2-binary/pip-egg-info/psycopg2_binary.egg-info
2020-10-09 02:06:45,187 ! writing /tmp/pip-install-5CmQNJ/psycopg2-binary/pip-egg-info/psycopg2_binary.egg-info/PKG-INFO
2020-10-09 02:06:45,188 ! writing top-level names to /tmp/pip-install-5CmQNJ/psycopg2-binary/pip-egg-info/psycopg2_binary.egg-info/top_level.txt
2020-10-09 02:06:45,188 ! writing dependency_links to /tmp/pip-install-5CmQNJ/psycopg2-binary/pip-egg-info/psycopg2_binary.egg-info/dependency_links.txt
2020-10-09 02:06:45,188 ! writing manifest file '/tmp/pip-install-5CmQNJ/psycopg2-binary/pip-egg-info/psycopg2_binary.egg-info/SOURCES.txt'
2020-10-09 02:06:45,189 ! 
2020-10-09 02:06:45,189 ! Error: pg_config executable not found.
2020-10-09 02:06:45,189 ! 
2020-10-09 02:06:45,190 ! pg_config is required to build psycopg2 from source.  Please add the directory
2020-10-09 02:06:45,190 ! containing pg_config to the $PATH or specify the full executable path with the
2020-10-09 02:06:45,190 ! option:
2020-10-09 02:06:45,191 ! 
2020-10-09 02:06:45,191 ! python setup.py build_ext --pg-config /path/to/pg_config build ...
2020-10-09 02:06:45,192 ! 
2020-10-09 02:06:45,192 ! or with the pg_config option in 'setup.cfg'.
2020-10-09 02:06:45,192 ! 
2020-10-09 02:06:45,193 ! If you prefer to avoid building psycopg2 from source, please install the PyPI
2020-10-09 02:06:45,193 ! 'psycopg2-binary' package instead.
2020-10-09 02:06:45,193 ! 
2020-10-09 02:06:45,194 ! For further information please check the 'doc/src/install.rst' file (also at
2020-10-09 02:06:45,194 ! <https://www.psycopg.org/docs/install.html>).
2020-10-09 02:06:45,194 ! 
2020-10-09 02:06:45,195 ! ----------------------------------------
2020-10-09 02:06:45,195 ! ERROR: Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.
```

it suggest me install `psycopg2`

scaytrase, 2020-10-08:
Solved by 
```
apt install libpq-dev python-pip
pip install psycopg2-binary
```

mpd352, 2020-10-09:
OllisGit:

Me again.  The 1.6.2 update hit today.  2 of the 3 instances I'm running upgraded fine.  One did not.

When it failed install from the plugin manager I tried installing it from the command line.  That failed, so I uninstalled the plugin and tried again.

That also failed from the manager and command line.

SOrry to bother you, but I'm just not that familiar with linuix to figure out the small details.  Log is attached.
[octoprint.log](https://github.com/OllisGit/OctoPrint-FilamentManager/files/5355533/octoprint.log)

Kevin

mpd352, 2020-10-09:
I resolved the issue through the back door.  I am running three instances of Octoprint. Two are connected to printers, the third is just a database server for filament manager. The filament manager instance and one of the printer instances are new, having been built up over the last 10 days in my fight to get filament manager to run under Python3.

Both of those instances successfully upgraded to 1.6.2. I just cloned the working printer instance, and reconfigured it, and now all three instances of Octoprint are connecting to the filament manager database.

Thanks, 

Kevin

OllisGit, 2020-10-10:
Glad to hear that you figured out the problem!

excalbian, 2020-10-12:
Just came to read this thread after the upgrade to 1.6.2 broke for me. Looks like most people have resolved this by installing pyscopg manually, so that's what I'm going to try next. I don't use the Postgres option for filament manager, so it seems a bit off that I have to install (at least part of?) Postgres to be able to use this plugin.

```
++++++++++++++++++++++++++++++++++++++
Now updating Filament Manager to 1.6.2
++++++++++++++++++++++++++++++++++++++
/home/pi/oprint/bin/python2 -m pip --disable-pip-version-check install https://github.com/OllisGit/OctoPrint-FilamentManager/releases/latest/download/master.zip --no-cache-dir
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting https://github.com/OllisGit/OctoPrint-FilamentManager/releases/latest/download/master.zip
Downloading https://github.com/OllisGit/OctoPrint-FilamentManager/releases/latest/download/master.zip (245kB)
Requirement already satisfied: OctoPrint in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint-FilamentManager==1.6.2) (1.4.2)
Requirement already satisfied: backports.csv<1.1,>=1.0.5 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint-FilamentManager==1.6.2) (1.0.7)
Requirement already satisfied: uritools<2.2,>=2.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint-FilamentManager==1.6.2) (2.1.1)
Requirement already satisfied: SQLAlchemy<1.2,>=1.1.15 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint-FilamentManager==1.6.2) (1.1.18)
Collecting psycopg2-binary
Downloading https://files.pythonhosted.org/packages/fc/51/0f2c6aec5c59e5640f507b59567f63b9d73a9317898810b4db311da32dfc/psycopg2-binary-2.8.6.tar.gz (384kB)
ERROR: Command errored out with exit status 1:
command: /home/pi/oprint/bin/python2 -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-3iEUIc/psycopg2-binary/setup.py'"'"'; __file__='"'"'/tmp/pip-install-3iEUIc/psycopg2-binary/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' egg_info --egg-base /tmp/pip-install-3iEUIc/psycopg2-binary/pip-egg-info
cwd: /tmp/pip-install-3iEUIc/psycopg2-binary/
Complete output (23 lines):
running egg_info
creating /tmp/pip-install-3iEUIc/psycopg2-binary/pip-egg-info/psycopg2_binary.egg-info
writing /tmp/pip-install-3iEUIc/psycopg2-binary/pip-egg-info/psycopg2_binary.egg-info/PKG-INFO
writing top-level names to /tmp/pip-install-3iEUIc/psycopg2-binary/pip-egg-info/psycopg2_binary.egg-info/top_level.txt
writing dependency_links to /tmp/pip-install-3iEUIc/psycopg2-binary/pip-egg-info/psycopg2_binary.egg-info/dependency_links.txt
writing manifest file '/tmp/pip-install-3iEUIc/psycopg2-binary/pip-egg-info/psycopg2_binary.egg-info/SOURCES.txt'
Error: pg_config executable not found.
pg_config is required to build psycopg2 from source.  Please add the directory
containing pg_config to the $PATH or specify the full executable path with the
option:
python setup.py build_ext --pg-config /path/to/pg_config build ...
or with the pg_config option in 'setup.cfg'.
If you prefer to avoid building psycopg2 from source, please install the PyPI
'psycopg2-binary' package instead.
For further information please check the 'doc/src/install.rst' file (also at
<https://www.psycopg.org/docs/install.html>).
----------------------------------------
ERROR: Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.
```