OllisGit/OctoPrint-FilamentManager #8: "Start Print" button disabled when the wrong spool is selected is not obvious

cyberjacob, 2020-10-04:
**Describe the bug**
When selecting a spool other than the one currently loaded, Filament Manager (correctly) does not allow the print to continue. However it is not obvious to the user why they cannot continue with the print.

**Steps to reproduce**
1. From the Filament Manager panel, select a spool loaded into a tool
2. Start a print
3. In the "confirm spools" modal, select a spool other than the one selected in step 1
4. "Start Print" button is disabled, with no obvious reasoning

**Expected behavior**
A message shows asking the user to confirm the loaded spools, and go back to correct Filament Manager if necessary. Alternatively, the modal could update the currently loaded spools, though this should probably have a confirmation tickbox or similar.

**Did the same happen when all other 3rd party plugins are disabled?**
N/A

**Log file**
N/A

**Screenshots**
Will add when this print has finished

**Additional context**
Add any other context about the problem here.

claricerogue, 2020-10-19:
I have the same issue but I noticed that "Tool" is not properly assigned. With some stls sliced with Cura, the Tool is changed to Tool 2 instead Tool 0, so I have to add two more extruders to the Octoprint Printer Config to be allowed to assign a Spool and then push the button to print. I can't find in the code what parameter or sentence can make this behavior so I'm afraid that is a Plugin problem.

DanHudson15, 2020-10-31:
This came up during a M600 filament change, and would not let me select a different spool, which I wanted to since I changed spools (color) on purpose.

Ajimaru, 2020-11-20:
Same behaviour for me, I cant print only with the first spool selected. I dont really understand the issue.
My printer has one extruder, why do I cant select a different spool for a certain job?

cyberjacob, 2020-11-20:
@Ajimaru I believe the intended operation is that you cancel the print from the confirmation dialogue, correct the setting in the widget on the main page, and then re-start the print

Ajimaru, 2020-11-25:
@cyberjacob yes you are right, I oversaw the widget on the main page.

cyberjacob, 2020-12-25:
As far as I'm aware, this is still outstanding

OllisGit, 2021-01-28:
Hi!

Well....it was designed by the origin developer like a "verification" dialog:
```
This dialog is meant to protect you from accidentally selecting wrong spools for the print.
```
Other tools use similar approaches: ```If you want to delete a github repository, enter the repo-name into the textfield ```

 ...there was already a discussion about this behavior in the old issue tracker.....this was one reason why I started developing my own filament/spool manager-plugin

Short: sorry about that behavior, but I am not willing to change that. You need to confirm the right spool or disable the dialog-box....or switch to SpoolManager ;-)

PS. Sorry, for being late to the party, but I made a "Octoprint sabatical" (no coding, no printing) the last months.

cyberjacob, 2021-01-29:
That sounds like a sensible decision, but would it not be worth having a message that tells the user they didn't select the right spool. Something like this?

    The spool you selected for hotend 1 differs from what Filament Manager expected. Please cancel this print, return to the main screen, and confirm the spool selected in the Filament Manager widget matches the spool currently loaded.

OllisGit, 2021-06-05:
fyi: In next release I will output the current selected spool in the confirm dialog. This was easy to do and help selecting the right spool.

OllisGit, 2021-06-05:
New Version 1.8.1 is out now

closed 2021-10-16 by OllisGit