OllisGit/OctoPrint-FilamentManager #10: Upgrade from 1.6.0 to 1.6.2 fails

hgroover, 2020-10-09:
**Describe the bug**
I am running Octoprint 1.4.2 via Octopi (installed recently). When trying to update to 1.6.2, I get this error:
(from octoprint.log)
2020-10-09 12:59:15,760 - octoprint.plugins.softwareupdate - INFO - Starting update of filamentmanager to 1.6.2...
2020-10-09 12:59:22,073 - octoprint.plugins.softwareupdate - ERROR - Update of filamentmanager can not be performed, please also check plugin_softwareupdate_console.log for possible causes of this
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/softwareupdate/__init__.py", line 1155, in _perform_update
    update_result = updater.perform_update(target, populated_check, target_version, log_cb=self._log, online=online)
  File "/home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/softwareupdate/updaters/pip.py", line 116, in perform_update
    raise exceptions.UpdateError("Error while executing pip install", (stdout, stderr))
UpdateError
2020-10-09 12:59:22,689 - octoprint.plugins.tracking - INFO - Sent tracking event update_failed, payload: {'to_version': '1.6.2', 'from_version': '1.6.0', 'target': u'filamentmanager'}

**Steps to reproduce**
1. Restart Octoprint
2. Check for update suggests a newer version of Filament Manager is available
3. Upgrade now
4. Failure as shown above. Tried a second time as other plugins were (successfully) updated, same result (after restarting Octoprint)

**Expected behavior**
Update should succeed.

**Did the same happen when all other 3rd party plugins are disabled?**
If not which plugin is causing the issue?

**Log file**
Drag and drop the octoprint.log here.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
Add any other context about the problem here.

Deses, 2020-10-09:
Yep. Can confirm.

cicinovec, 2020-10-09:
I also confirm this bug

plorimer, 2020-10-09:
Well, good to know it's nothing on my end causing this.

mpd352, 2020-10-09:
This is probably going to help the rest of you guys, but I did resolve my issue with a failed upgrade to 1.6.2.

I resolved the issue through the back door.  I am running three instances of Octoprint. Two are connected to printers, the third is just a database server for filament manager. The filament manager instance and one of the printer instances are new, having been built up over the last 10 days in my fight to get filament manager to run under Python3.

Both of those instances successfully upgraded to 1.6.2. I just cloned the working printer instance, and reconfigured it, and now all three instances of Octoprint are connecting to the filament manager database.

Thanks, 

Kevin

JamieS01, 2020-10-09:
Same problem here 👍

hgroover, 2020-10-09:
OK, this worked for me on Octopi - I'm in the middle of a very long print job so will not be able to try the actual upgrade anytime soon (about 16 hours to go):

1. ssh into pi
2. sudo apt-get update
3. PATH=/home/pi/oprint/bin:${PATH}
4. pip install psycopg2-binary
(just to show you that it will fail in a similar manner to the error)

Now we'll fix the missing pg_config:
5. **sudo apt-get install libpq-dev**
(thank you StackOverflow)

And now the pip install will work:
6. pip install psycopg2-binary

There will be some warnings about upgrading pip and the upcoming deprecation of Python 2.7 but I believe this will solve the problem. I will not be able to verify for some hours.

hgroover, 2020-10-10:
I was successfully able to update. I don't know if the pip install was needed but the essential step was

sudo apt-get install libpq-dev

thegarbz, 2020-10-10:
Can confirm @hgroover solution worked. 
Note: apt-get install libpq-dev initially failed. I had to run "sudo apt update" first.

hgroover, 2020-10-10:
> Can confirm @hgroover solution worked.
> Note: apt-get install libpq-dev initially failed. I had to run "sudo apt update" first.

Thanks, updated the issue. Good to know the pip command wasn't also needed.

12345Josparky, 2020-10-10:
Ok so I don't know how to do all those commands but mine still is not updating through the update manager

hgroover, 2020-10-10:
@12345Josparky that's just a workaround for those who have ssh / sudo access (not that hard but there are a few steps you might have to go through, like setting up putty or some other ssh client, and knowing the password used to setup your installation of Octopi, assuming that's what you're using; but if you're not already experienced at this it's better to wait for an update).

The maintainers of this addon will probably come out with an update soon that will address this. I'm just a user who is grateful to have this very useful addon and has provided detailed info on what happened and what fixes it. My best suggestion is to wait, seeing that the maintainers here are pretty active they will have an update before long. In the meantime the previous version should work perfectly well.

I hope that helps!

12345Josparky, 2020-10-10:
Thanks for the quick reply I do have putty but I am not that familiar with it at all lol. I watched a video that showed me how to set it all up and that was couple years ago. I don't even remember how I done it lol. I can wait no biggy. Thanks for your help and reply.

spanzetta, 2020-10-11:
I also confirm it works with the manual update

maxmednik, 2020-10-11:
Same problem here. Can we just not worry about manually updating and wait until a future update fixes this so it can auto-update via Octoprint by itself smoothly?

thegarbz, 2020-10-11:
Sure but I wouldn't hold your breath. The problem appears to be in pip not in this plugin. Outdated versions of pip fail to install the dependency. You will likely need to wait for Octopi itself to receive an update.

Deses, 2020-10-11:
> Same problem here. Can we just not worry about manually updating and wait until a future update fixes this so it can auto-update via Octoprint by itself smoothly?

If you want to keep receiving the update notification, sure. 

`sudo apt-get install libpq-dev`, it's not that hard.

maxmednik, 2020-10-12:
Is it just that one command? I see.5 other steps in hgroover's reply earlier, and I'm worried about one of those failing also. Should I try just the one command you wrote, Deses, or all 6 of hgroover's commands?

hgroover, 2020-10-12:
No, you just need the two, sudo apt update and sudo apt install. I revised
the original ticket.

None of those additional steps will hurt but you definitely don't need
them. I was just showing my process. As it turns out many people have
already confirmed you just need the two commands.

The point about this being a pip issue due to using the deprecated Python
2.7 (rather than Python 3) is interesting. This workaround may have broader
use than just for this plugin.

12345Josparky, 2020-10-12:
so where do you put the commands can they be put in octoprint??

mpd352, 2020-10-12:
download and install PuTTY.  https://www.chiark.greenend.org.uk/~sgtatham/putty/latest.html.

use the ip address of your octoprint instance to telnet direct to it witH PuTTY.  log in with the username and password you used to create your octoprint instance, then enter the commands listed.

DeviousPenguin, 2020-10-12:
Thanks for posting the fix, `sudo apt install libpq-dev` worked great for me :)

thegarbz, 2020-10-12:
@12345Josparky if you have a keyboard and a TV it may be easier to simply plug the Raspberry Pi into those, login on the console and then type those commands. If you've not used SSH before this may be easier. 

Otherwise download putty.
Under Hostname / IP on the first window put in "octopi.local" and click Open
Default username: pi
Default password: raspberry. 
Type: sudo apt update (when prompted give that default password again).
Type: sudo apt-get install libpq-dev
Type: exit.

And you should be all done. I can't remember if the default octopi image has ssh enabled though or if I took additional steps to make this work for me. But as said above the same stuff works by logging directly into the console, all you need is a keyboard and a hdmi cable.

12345Josparky, 2020-10-12:
Thank you all for the detailed instructions. I do have one more question. May be a dumb one lol. I am assuming I need to download the update from here and have it somewhere so it can install it. I appreciate all you guys help. I do have putty installed on my pc just not familiar with it. I watched a video to get it all installed before.

hgroover, 2020-10-12:
Not a dumb question by any means. After doing the apt update / apt install (assuming it completes successfully) just go back to Octoprint / settings / updates and click on the update button for Filament Manager. This time it should succeed.

12345Josparky, 2020-10-12:
So if I can ask one more question so I understand this a little more for future. What is this updating exactly? And thank you all for being so nice to help the ones that don't know that much about putty. I am always wanting to learn more.

12345Josparky, 2020-10-12:
Thanks hgroover

thegarbz, 2020-10-12:
Yeah @12345Josparky to answer your question specifically on what this did:
In the release notes of FilamentManager you can see 1.6.2 "_#7 added postgressql database driver to the plugin_" 

When the update runs you can see above the error message says 
"_Error: pg_config executable not found. pg_config is required to build psycopg2 from source_." This is pip (Package Installer for Python) trying and failing to install the postgressql database driver library that this new version depends on. Rather than take the drastic method of updating pip (which according to a quick internet search would fix this, but a task best left to the octodash team), we instead are installing the package libpq-dev which provides the missing file from the operating system repository via Aptitude. 

So what we did with the commands:
_sudo_ - this tells Linux to run the following commands with elevated privileges. Aptitude only runs as root. 
_apt update_ - this tells apt to go back to the repository from Raspbian and get a list of all software available, its dependencies, and its current versions. You may have gotten a message after running this that 100 updates are available or something like that. We simply ignored that.
_apt-get install libpq-dev_ - this should be self explanatory, it fetches and installs the libpq-dev library which includes the missing pg_config.

With the missing software installed, pip is able to download and install psycopg2 and then install this plugin.

OllisGit, 2020-10-12:
Hi all,
I am really sorry, that you have problems to install the latest version which includes PostgreSQL-Driver.

I thought it is just an other dependency which I could add to the requirement section in the plugin.

Now, I removed the driver from the plugin (Version 1.6.3), If someone needs to use the external database, he/she can look into the wiki-page: https://github.com/OllisGit/OctoPrint-FilamentManager/wiki/Setup-PostgreSQL-on-Raspbian-(Stretch)

thegarbz, 2020-10-12:
@OllisGit it should have been. Maybe something to do with python version being outdated. At least on the psycopg github page there are people saying an updated version of pip fixes the problem, and frankly it should since the whole point of the binary package is that it shouldn't need to be built. https://github.com/psycopg/psycopg2/issues/699

hgroover, 2020-10-13:
@OllisGit I can confirm that the upgrade from 1.6.0 to 1.6.3 went smoothly on another Octopi installation (where I had not applied the apt workaround). Can we close this issue now? If you'd rather I do so please let me know... Thank you!

closed 2020-11-22 by stale[bot]