OllisGit/OctoPrint-FilamentManager #11: Unable to install plugin via Plugin Manager on Octopi

joshuapeterson, 2020-10-10:
**Describe the bug**
Using the newest version of octopi downloaded from octoprint.org (0.17.0), I am unable to install FilamentManager for the plugin manager.

**Steps to reproduce**
1. Setup raspberry pi with Octopi 0.17.0
2. Click on plugin manager in settings and Get More...
3. Search for FilamentManager
4. Click Install
5. See this error
ERROR: Command errored out with exit status 1:
command: /home/pi/oprint/bin/python2 -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-tvBWos/psycopg2-binary/setup.py'"'"'; __file__='"'"'/tmp/pip-install-tvBWos/psycopg2-binary/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' egg_info --egg-base /tmp/pip-install-tvBWos/psycopg2-binary/pip-egg-info
cwd: /tmp/pip-install-tvBWos/psycopg2-binary/
Complete output (23 lines):
running egg_info
creating /tmp/pip-install-tvBWos/psycopg2-binary/pip-egg-info/psycopg2_binary.egg-info
writing /tmp/pip-install-tvBWos/psycopg2-binary/pip-egg-info/psycopg2_binary.egg-info/PKG-INFO
writing top-level names to /tmp/pip-install-tvBWos/psycopg2-binary/pip-egg-info/psycopg2_binary.egg-info/top_level.txt
writing dependency_links to /tmp/pip-install-tvBWos/psycopg2-binary/pip-egg-info/psycopg2_binary.egg-info/dependency_links.txt
writing manifest file '/tmp/pip-install-tvBWos/psycopg2-binary/pip-egg-info/psycopg2_binary.egg-info/SOURCES.txt'
Error: pg_config executable not found.
pg_config is required to build psycopg2 from source.  Please add the directory
containing pg_config to the $PATH or specify the full executable path with the
option:
python setup.py build_ext --pg-config /path/to/pg_config build ...
or with the pg_config option in 'setup.cfg'.
If you prefer to avoid building psycopg2 from source, please install the PyPI
'psycopg2-binary' package instead.
For further information please check the 'doc/src/install.rst' file (also at
<https://www.psycopg.org/docs/install.html>).
----------------------------------------
ERROR: Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.
Error!
Could not parse output from pip, see plugin_pluginmanager_console.log for generated output

**Expected behavior**
FilamentManager to be installed

**Did the same happen when all other 3rd party plugins are disabled?**
If not which plugin is causing the issue?

**Log file**
Drag and drop the octoprint.log here.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
Add any other context about the problem here.

gdombiak, 2020-10-10:
Got same error updating to latest version of the plugin

vr6syncro, 2020-10-10:
same here on a pi4

Syl20-94, 2020-10-10:
Hello,
same for me, on Octoprint 0.17 with FilamentManager 1.6.2, on Rasp4. til now update was OK - meaning 1.6.1 is running well

I've got this log  from plugin_octoprint.log 
 Downloading psycopg2-binary-2.8.6.tar.gz (384 kB)
2020-10-09 12:24:01,262 ! ERROR: Command errored out with exit status 1:
2020-10-09 12:24:01,264 ! command: /home/pi/oprint/bin/python2 -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-D2cDjz/psycopg2-binary/setup.py'"'"'; __file__
='"'"'/tmp/pip-install-D2cDjz/psycopg2-binary/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(comp
ile(code, __file__, '"'"'exec'"'"'))' egg_info --egg-base /tmp/pip-pip-egg-info-0SNUMd
2020-10-09 12:24:01,264 ! cwd: /tmp/pip-install-D2cDjz/psycopg2-binary/
2020-10-09 12:24:01,265 ! Complete output (23 lines):
2020-10-09 12:24:01,265 ! running egg_info
2020-10-09 12:24:01,265 ! creating /tmp/pip-pip-egg-info-0SNUMd/psycopg2_binary.egg-info
2020-10-09 12:24:01,266 ! writing /tmp/pip-pip-egg-info-0SNUMd/psycopg2_binary.egg-info/PKG-INFO
2020-10-09 12:24:01,266 ! writing top-level names to /tmp/pip-pip-egg-info-0SNUMd/psycopg2_binary.egg-info/top_level.txt
2020-10-09 12:24:01,267 ! writing dependency_links to /tmp/pip-pip-egg-info-0SNUMd/psycopg2_binary.egg-info/dependency_links.txt
2020-10-09 12:24:01,267 ! writing manifest file '/tmp/pip-pip-egg-info-0SNUMd/psycopg2_binary.egg-info/SOURCES.txt'
2020-10-09 12:24:01,267 !
2020-10-09 12:24:01,268 ! Error: pg_config executable not found.
2020-10-09 12:24:01,268 !
2020-10-09 12:24:01,269 ! pg_config is required to build psycopg2 from source.  Please add the directory
2020-10-09 12:24:01,269 ! containing pg_config to the $PATH or specify the full executable path with the
2020-10-09 12:24:01,269 ! option:
2020-10-09 12:24:01,270 !
2020-10-09 12:24:01,270 ! python setup.py build_ext --pg-config /path/to/pg_config build ...
2020-10-09 12:24:01,270 !
2020-10-09 12:24:01,271 ! or with the pg_config option in 'setup.cfg'.
2020-10-09 12:24:01,271 !
2020-10-09 12:24:01,271 ! If you prefer to avoid building psycopg2 from source, please install the PyPI
2020-10-09 12:24:01,272 ! 'psycopg2-binary' package instead.

FrancYescO, 2020-10-10:
confirmed

LordMM74, 2020-10-10:
I had same problem. Tried to upgrade to latest version on an raspberr pi 3b+.

FrancYescO, 2020-10-10:
probably the workaround is here https://github.com/OllisGit/OctoPrint-FilamentManager/issues/10#issuecomment-706502123

thegarbz, 2020-10-10:
Hi all, solution in the #10 works.

Run: 
"sudo apt update" 
"sudo apt-get install libpq-dev"

This seems to be related to outdated versions of pip not autobuilding the pq library correctly.

Syl20-94, 2020-10-10:
hi,
yep the workaround is working with the install of libpq-dev
thank a lot

joshuapeterson, 2020-10-10:
That workaround worked for me!  Thank you!

OllisGit, 2020-10-12:
Hi all,
I am really sorry, that you have problems to install the latest version which includes PostgreSQL-Driver.

I thought it is just an other dependency which I could add to the requirement section in the plugin.

Now, I removed the driver from the plugin (Version 1.6.3), If someone needs to use the external database, he/she can look into the wiki-page: https://github.com/OllisGit/OctoPrint-FilamentManager/wiki/Setup-PostgreSQL-on-Raspbian-(Stretch)

closed 2020-11-21 by stale[bot]