OllisGit/OctoPrint-FilamentManager #12: Add notes to Spool

JcMinarro, 2020-10-16:
**Describe the solution you'd like**
When I configure a new spool, I would like to add some notes to it.
Would be great to have an input box where we can write some text.
Some example could see store info about the ideal temperature or any other information we want to store related with the filament

**Describe alternatives you've considered**
Add an extra input box on the Spool Popup that allow us insert some notes

OllisGit, 2020-11-03:
Please take a look to my other plugin: https://github.com/OllisGit/OctoPrint-SpoolManager
There you can add notes to each spool.

JcMinarro, 2020-11-03:
Hello  @OllisGit 
Are they complementary? 
I mean, I have already configured all my Spools on the previous plugin, if I install this new one, would I get the information from the other plugin?
Will they be updated simultaneously?

OllisGit, 2021-01-12:
Hi @JcMinarro,

you can install both plugins in parallel. Each plugin use it's own database. There is no synching between the two plugins.

If you install SpoolManager you can export the old database from FilamentManager as a CSV, cleanup or add more informations (like notes) and import the CSV in the SpoolManager-Database.

PS. Sorry, for being late to the party, but I made a "Octoprint sabatical" (no coding, no printing) the last months.

closed 2021-01-28 by OllisGit