OllisGit/OctoPrint-FilamentManager #13: 2-spool empty warning

erichiller, 2020-10-21:
My printer has dual extruders & filament spools. If one filament is empty, I get a notice about not enough filament, even when I'm not using that spool.

**Steps to reproduce**
1. With a two spool printer have one spool set to 0 remaining
2. Print something
3. Observe notice in top right
> Insufficient filament
> The current print job needs more material than what's left on the selected spool.

**Expected behavior**
Only error when using the spool that is empty.

**Notice**
[image]

**Selected Spools**
[image]

**First Spool & spool in use**
[image]

**Second spool (empty)**
[image]

OllisGit, 2021-01-12:
Hi @erichiller,

because I don't have a dual extruder printer, please provide your gcode so that I can try to reproduce the behaviour with a "virtual printer".

PS. Sorry, for being late to the party, but I made a "Octoprint sabatical" (no coding, no printing) the last months.

OllisGit, 2021-04-19:
no feedback -> closed

closed 2021-04-19 by OllisGit