OllisGit/OctoPrint-FilamentManager #14: Wrong temperature offset

KZL1992, 2020-10-22:
Hello, I have been using FilamentManager and setup a spool using an offset of -20 degree, but it seems that 
my printer is running at 190 instead of 180 from a gcode set at 200

1. Go to setttings
2. Add new pool: (Eryone - Glitter Black) with temperature offset at -20
3. Run your print
4. Temperature is at 190

[octoprint.log](https://github.com/OllisGit/OctoPrint-FilamentManager/files/5421865/octoprint.log)

TheRealAlexV, 2020-11-29:
I've also experienced similar behavior.

Ajimaru, 2021-01-03:
Filament Temeprature offset is not working in my installation too.

OllisGit, 2021-01-28:
Hi @xKZL,

in your log file I saw a lot of errors from other plugins, but also from filamentmanager 
```
2020-10-21 19:22:17,925 - octoprint.plugin.core - ERROR - Error loading plugin filamentmanager
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/plugin/core.py", line 972, in _import_plugin
    module = _load_module(module_name, spec)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/plugin/core.py", line 71, in _load_module
    return imp.load_module(name, f, filename, details)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_filamentmanager/__init__.py", line 17, in <module>
    from .data import FilamentManager
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_filamentmanager/data/__init__.py", line 11, in <module>
    from backports import csv
ImportError: No module named backports
```
...but behind that error you restarted the server again and the error-message was not shown. 

Are you using an external DB?

@all
I could not reproduce the issue on my system. I used the following test gcode:
```
; Temperature Test
M140 S60
M105
M190 S60
M104 S200
M105
M109 S200
M82 ;absolute extrusion mode
```
Result: 
[image]
Please try yourself with the test code and/or look into your gcode, maybe there is a temp assignment that you didn't expect

NotExpectedYet, 2021-02-15:
I'm having OctoFarm users telling me of this issue. Not tested it personally as I'm having a printing sabatical myself aha. I've shared this issue with them so hopefully they will come respond.

closed 2021-04-19 by OllisGit