OllisGit/OctoPrint-FilamentManager #16: Filament Manager does not track filament use with multiple extruders.

mpd352, 2020-11-05:
I recently added a multiple extruder printer, a GeeeTech A10T.  I've set up OctoPrint for three extruders, and I can control the extruders from Octoprint.  Filament Manager allows me to select a spool for each of the three extruders. However, Filament Manager is not tracking filament use on any of the schools I've selected.

I see this error in the log:  2020-11-04 11:34:04,021 - octoprint.plugins.filamentmanager - ERROR - Failed to update selected spool for tool0: 'NoneType' object has no attribute 'update_selection'

[octoprint-11-04-2020.log](https://github.com/OllisGit/OctoPrint-FilamentManager/files/5491286/octoprint-11-04-2020.log)

As I've asked you in the past, Oli, what am I doing wrong?

Thanks for your help, as usual.

Kevin

OllisGit, 2021-01-12:
Hi @mpd352 ,

you mentioned error has nothing to do with the issue that the filemament is not tracked.
At 11:34:04 the filamentmanger-plugin is not initialized (starting: 11:34:24), but the browser is open and tries to call the filamentmanager-api. I will change the log behaviour in the next version.

But I can see that the tracking for all tools is done:
2020-11-04 11:49:43,380 - octoprint.plugins.filamentmanager - DEBUG - Updated thresholds: {'tool0': 77295.003846973, 'tool1': 76065.70925566753, 'tool2': 332501.35046143027}

Is this maybe only a UI-Bug. So the data is not transfered to the UI?

Can you please post a screenshot and the log after a print? Please also mention what you filament amount do you except.

PS. Sorry, for being late to the party, but I made a "Octoprint sabatical" (no coding, no printing) the last months.

mpd352, 2021-01-12:
OllisGit:

I will print a piece and post screen shots and logs as soon as I can.  I've another issue right now.  Since the last OctoPrint update, when I try to print to the A10T with Octoprint I get non-stop Invalid Tool messages.  I'm in the process of trying again right now to collect logs and screen shots and post them to the forum.

Once that issue is resolved I'll revisit the filament tracking issue and send you logs and screen shots.

Thanks

mpd352, 2021-01-17:
OllisGit:

The printer is printing successfully with Octoprint again.  It was a Simplify3D issue, not OctoPrint.  The issue of not tracking multiple spools still remains.  The logs and screen shots are included.

Spool counts before the print:
[image]

 Spool counts after the print:
[image]

What Octoprint indicates the filament usage should be:
[image]

Octoprint Log:
[octoprint-Conde-01-17-21.log](https://github.com/OllisGit/OctoPrint-FilamentManager/files/5826965/octoprint-Conde-01-17-21.log)

Serial Log:
[serial-Conde-01-17-21.log](https://github.com/OllisGit/OctoPrint-FilamentManager/files/5826967/serial-Conde-01-17-21.log)

Thanks,

Kevin

OllisGit, 2021-01-28:
Hi @mpd352,
thx a lot for the screenshot really helpful, but can you also provide me the corresponding gcode?

OllisGit, 2021-01-29:
Hi ,

new release 1.7.1 is out and should fix some material tracking issues.
After resuming a multi extruder print, the following tracked filament is assigned to tool0. To fix that issue I reimplemented the whole material tracking (maybe it solves other issues as well ;-). 

Please retest with the new version and give me a  feedback!

mpd352, 2021-01-29:
OllisGit:

I installed the update successfully on one machine, but when I installed it on the A10T, which is also my Postgres machine, FIlament Manager disappeared from the Plugin Manager, and it no longer shows up in settings.  It also is missing from the plugins list.  Originally it also did not appear in the add plugin dialog, so I cut and pasted the install url manually.  The log showed it installed, but it still does not show up in the Plugin Manager or plugins list.

Here's a screen shot and the logs:

[image]
[octoprint.log](https://github.com/OllisGit/OctoPrint-FilamentManager/files/5896695/octoprint.log)
[plugin_pluginmanager_console.log](https://github.com/OllisGit/OctoPrint-FilamentManager/files/5896696/plugin_pluginmanager_console.log)
[plugin_softwareupdate_console.log](https://github.com/OllisGit/OctoPrint-FilamentManager/files/5896697/plugin_softwareupdate_console.log)
[serial.log](https://github.com/OllisGit/OctoPrint-FilamentManager/files/5896698/serial.log)

I can't test the update until I can get it working.  Is there a way to manually remove Filament Manager, so I can do a clean reinstall?  If I do would it require a new Postgres install as well?

mpd352, 2021-01-30:
My other instances Filament Manager running on my two there Octoprint servers are working and attaching to the Postgres DB on  the same OctoPrint server that is mentioned above.

OllisGit, 2021-01-31:
Hi @mpd352,
sorry, but Version 1.7.1 is broken, please use Version [1.7.2](https://github.com/OllisGit/OctoPrint-FilamentManager/releases/tag/1.7.2)

mpd352, 2021-01-31:
OllisGit:

It installed.  It's running a multicolor job right now.  I'll let you know if it tracks usage on all three spools.

Thanks for all the hard work!

Kevin

mpd352, 2021-02-01:
The issue is fixed!  Bug squashed!  Thank you!

closed 2021-02-01 by mpd352