OllisGit/OctoPrint-FilamentManager #17: Does not allow spool change on GCODE filament change (PrusaSlicer)

bovlb, 2020-11-07:
**Describe the bug**

I often do multi-filament prints, and these are not accounted correctly by FilamentManager because there is no opportunity to change spool mid-print.

**Steps to reproduce**
1. Slice a model in PrusaSlicer
2. Set a filament change
3. Generate GCODE
4. Send GCODE to printer
5. Wait for printer to request filament change
5. Try to change spool in Filament Manager

**Expected behavior**
The ideal behaviour would be for FilamentManager to prompt me for a new spool selection when the printer requests a filament change.
An alternative solution would be if FilamentManager allowed me to change filament mid-print.

**Did the same happen when all other 3rd party plugins are disabled?**
n/a

**Log file**
n/a

**Screenshots**
n/a

**Additional context**
See also:
https://github.com/malnvenshorn/OctoPrint-FilamentManager/issues/182
https://github.com/malnvenshorn/OctoPrint-FilamentManager/issues/160
https://github.com/malnvenshorn/OctoPrint-FilamentManager/issues/76

bovlb, 2020-12-07:
I have found a workaround, but it's not pretty:
1. Wait for printer to request filament change
2. Hit Pause button on Octoprint UI
3. Change physical filament in the usual way
4. Wait while printer moves into position to continue and then executes pause
5. Change virtual filament in UI (possible because paused)
6. Hit Resume button in Octoprint UI
6. Printer resumes printing with new filament

OllisGit, 2021-01-12:
Hi @bovlb,

I added this M600 feature to my SpoolManager-Roadmap: https://github.com/OllisGit/OctoPrint-SpoolManager/projects/1
Issue: https://github.com/OllisGit/OctoPrint-SpoolManager/issues/34

So, I will not implement this feature in the FilamentManager-Plugin, sorry!

PS. Sorry, for being late to the party, but I made a "Octoprint sabatical" (no coding, no printing) the last months.

closed 2021-01-28 by OllisGit