OllisGit/OctoPrint-FilamentManager #19: No spool selection after reboot

taker218, 2020-11-13:
**Describe the bug**
When I reboot the OctoPi instance (complete reboot, not just restart) I can't select any spool. I am using an external database for my instances and this is on every one of them.
I know this might not be completely bound to this plugin, but maybe you have an idea how to work around this problem.

Once I restart octoprint (via webinterface) the selection is possible.
It might have something to do with the way octoprint is starting. I don't know if the octoprint service is waiting for the network to come up. And as far as I know from the old plugin author, the plugin only checks on startup of the instance, if the remote database is available. If not, you don't get a selection.

**Steps to reproduce**
1. set up filament manager with external database on another machine
2. reboot your octoprint device
3. after reboot no selection is possible

**Expected behavior**
A clear and concise description of what you expected to happen.

**Did the same happen when all other 3rd party plugins are disabled?**
If not which plugin is causing the issue?

**Log file**
Drag and drop the octoprint.log here.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
Add any other context about the problem here.

msEmmaMays, 2020-12-04:
This happened to me because of the UI customizer plugin hiding it.   I had open it's settings and hit save and apparently that defaulted to not showing the plugin for some reason.   Perhaps someone should file a bug report over there?

[image]

taker218, 2020-12-07:
I think that's another problem @msEmmaMays 

I can see the plugin and everything, but I can't select a spool. It doesn't show any spool to select, but the selection box is there.
I know this is a problem of how the plugins are able to check for a network connection and I already tried to start octoprint after the network is online, but this didn't work. Octoprint would start at all like that.

OllisGit, 2021-01-12:
Hi @taker218,

is the selection box empty or not possible to select a single spool? Or is the box "greyed out" (disabled). 
Reason for disabled: printer state = printing or user is not logged in

please try the following:
- Disable all "UI related" Plugins, like the mentioned "UI customizer"
- Open the Browser Developer Tools (press F12), log for error messages
- Look also into the network-tab of the browser developer tools

taker218, 2021-01-14:
Hi,

the selection box was empty, but right now I can't recreate the problem on any of my installations.
I thought it was a a problem related to python 2 because I can't recall the problem to have happened with my python3 instances. But right now it doesn't happen any more.

taker218, 2021-01-14:
Okay, I just got it to reproduce.

I updated one instance to python 3 and now I get the error, but also when I restart Octoprint.
I attached the error from the console and from the network tab.
[console-export-2021-1-14_16-5-50.txt](https://github.com/OllisGit/OctoPrint-FilamentManager/files/5815255/console-export-2021-1-14_16-5-50.txt)
[network-export-2021-1-14.txt](https://github.com/OllisGit/OctoPrint-FilamentManager/files/5815256/network-export-2021-1-14.txt)

closed 2021-01-28 by OllisGit