OllisGit/OctoPrint-FilamentManager #21: Spool selection when print starts

taker218, 2020-11-24:
**Describe the bug**
I select the currently loaded spool in the widget and when I start the print, the dialogue comes up to confirm the selection. I would have thought that the currently selected spool would be the standard one in the dialogue. But I have to select this spool everytime I start a print again and can't just confirm the dialogue.

**Steps to reproduce**
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Did the same happen when all other 3rd party plugins are disabled?**
If not which plugin is causing the issue?

**Log file**
Drag and drop the octoprint.log here.

**Screenshots**
[image]
[image]

**Additional context**
Add any other context about the problem here.

Ajimaru, 2020-11-25:
Hello @taker218 , what you discribe is not a bug, see #8 
[image]

taker218, 2020-11-25:
I think the issue in #8 is different, since it's about changing the spool in the selection before the print.

The dialogue before the print should have the spool, which is selected in the widget, as default selection.
At least that's what I was expecting as a default behaviour.

Ajimaru, 2020-11-25:
If the dialogue behaves how you describe then the dialogue is redundant. As you know user unintentionally not read message boxes.
It make sense that the user actively have to choose the correct filament again in this conformation dialogue.

closed 2021-01-04 by stale[bot]