OllisGit/OctoPrint-FilamentManager #23: Plugin doesn't install in Octoprint / Windows / Python 3.9 - time.clock() deprecated

SebMeunier, 2020-11-27:
**Describe the bug**
The plugin doesn't install from Octoprint UI in Windows, with Python 3.9. 
Trying to install it manually gives a success, but Plugin isn't present in Octoprint. 

**Steps to reproduce**

See all details here: https://community.octoprint.org/t/filamentmanager-plugin-install-issue-windows/26718/3

**Expected behavior**
Installation working :-)

**Did the same happen when all other 3rd party plugins are disabled?**
Yes

**Log file**
[octoprint.log](https://github.com/OllisGit/OctoPrint-FilamentManager/files/5609676/octoprint.log)

**Screenshots**
See https://community.octoprint.org/t/filamentmanager-plugin-install-issue-windows/26718/3

**Additional context**

SebMeunier, 2020-12-28:
@OllisGit any chance to get an update from you about this?

Micha7719, 2020-12-30:
same Problem here. any news?

felda, 2021-01-07:
I am also having this issue on Windows 10, python 3.9.

OllisGit, 2021-01-14:
PS. Sorry, for being late to the party, but I made a "Octoprint sabatical" (no coding, no printing) the last months.

Will check if it possible to upgrade the sql-dependencies.

Sigilos, 2021-02-06:
Having this same issue, but accessing the Octopi UI from Chromebook instead of Win 10. No matter how many reboots or installs from the plugin manager, this plugin just won't show up in the UI or the list of Installed Plugins.

OllisGit, 2021-02-07:
It doesn't look so good, it will take a lot of effort to make the plugin Python 3.8+ ready....maybe I can just upgrade the SQL-Library and release this version as an untested version.

cp2004, 2021-02-07:
To solve this issue *all* you have to do is upgrade the SQLAlchemy dependency. Emphasis on *all* because of course that is easier said than done, since the API might have changed. I just checked, the latest version (1.3.23) is still compatible with Python 2 and 3.4+, so it would work for that.

cp2004, 2021-02-07:
It's been a long time since I have used SQL Alchemy, but it is a very popular library/framework so I wouldn't expect too much of an issue - it is only one minor release 'jump'. Since it is so heavily depended on, I don't imagine they would break things that easily.

lmerega, 2022-10-06:
Hallo, any news about this issue? I cannot use filament manager on my Windows Instllation and it would be really nice.