OllisGit/OctoPrint-FilamentManager #24: FM on tracks filament use on Spool o

mpd352, 2020-12-03:
**Describe the bug**
A clear and concise description of what the bug is.
I have a GeeeTech A10T, which is a three spool printer. Filament Manager is configured to allow me to select a specific spool from the spool database for each extruder. The correct number of extruders is set in Octoprint Setup, and the shared hotend flag is set.

**Expected behavior**
Filament usage should be tracked on Spool 1 and 2 as well

**Log file**
Drag and drop the octoprint.log here.
[octoprint Part 01.log](https://github.com/OllisGit/OctoPrint-FilamentManager/files/5638410/octoprint.Part.01.log)
[octoprint Part 02.log](https://github.com/OllisGit/OctoPrint-FilamentManager/files/5638411/octoprint.Part.02.log)

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]
[image]
[image]

**Additional context**
Add any other context about the problem here.

WarrenSchultz, 2021-01-14:
Confirmed, seeing this issue as well, came here to file a bug. :)

OllisGit, 2021-01-28:
As mentioned in the other ticket please provide the corresponding gcode file, so that I am able to reproduce the issue in my environment.
@WarrenSchultz same question to you:
- screenshot of what you expect and the gcode file.

mpd352, 2021-01-28:
Sorry, thought I had uploaded these. 

[Ornaments 01_colored.txt](https://github.com/OllisGit/OctoPrint-FilamentManager/files/5890007/Ornaments.01_colored.txt)
[image]

The forum won't let me upload a gcode file so I renamed Ornaments.01_colored.gcode to Ornaments.01_colored.txt

OllisGit, 2021-01-28:
Hi @mpd352 ,
I looked into the gcode and there is only T2 used and not T1.
[image]

This is also detected by OctoPrint itself, only Tool 2 is shown:
[image]

WarrenSchultz, 2021-01-28:
@OllisGit 
[image]
[LeftAnchorBlock_0.2mm_ABS_MK3SMMU2S_12h19m.gcode.zip](https://github.com/OllisGit/OctoPrint-FilamentManager/files/5890605/LeftAnchorBlock_0.2mm_ABS_MK3SMMU2S_12h19m.gcode.zip)

[image]

The print is set to use Tool 4, the Filament Manager has Tool 4 showing the correct filament. Printing, everything works well, but the remaining filament counter for tool 4 (in this case, ABS) is not decremented with use. On a print using T0 (PLA in this case), it would work fine.

OllisGit, 2021-01-29:
Hi @WarrenSchultz,

I did some test and for your provided gcode the FilamentManager only detect ~55mm for Tool4, but it should 76997.75167999369.

Currently I am rewriting the whole filamenttracker-logic, because it didn't work in some cases, "obviously" ;-)
Will be finished this weekend...hopefully

mpd352, 2021-01-29:
> Hi @WarrenSchultz,
>
> I did some test and for your provided gcode the FilamentManager only detect ~55mm for Tool4, but it should 76997.75167999369.
>
> Currently I am rewriting the whole [...]

Thank you!  We appreciate the time and effort.

OllisGit, 2021-01-29:
Hi,

new release 1.7.1 is out and should fix some material tracking issue.
After resuming a multi extruder print, the following tracked filament is assigned to tool0. To fix that issue I reimplemented the whole material tracking (maybe it solves other issues as well ;-). 

Please retest with the new version and give me a  feedback!

closed 2021-04-19 by OllisGit