OllisGit/OctoPrint-FilamentManager #26: Update Filament Manager on Octoprint 1.5.0 not possible 404 File Not Found

becksteiner, 2020-12-04:
**Describe the bug**
It is starting to update the Filament Manager but when trying to download the files from git it's showing the error 404 Not Found.

**Steps to reproduce**
1. Go to 'Octoprint Settings'
2. Click on 'Software Update'
3. Scroll down to 'Filament Manager: 0.5.3 '
4. Click Update
5. Pop Up is showing the update Info > Click on continue
6. Error

**Expected behavior**
Not Error -> Update Succesfull and restart of octoprint

**Did the same happen when all other 3rd party plugins are disabled?**
I see no relation to the error message.

**Log file**
Aktualisiere gerade, bitte warten.
+++++++++++++++++++++++++++++++++++++++
Aktualisiere Filament Manager auf 1.6.3
+++++++++++++++++++++++++++++++++++++++
/home/pi/oprint/bin/python2 -m pip --disable-pip-version-check install https://github.com/malnvenshorn/OctoPrint-FilamentManager/archive/1.6.3.zip --no-cache-dir
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting https://github.com/malnvenshorn/OctoPrint-FilamentManager/archive/1.6.3.zip
ERROR: HTTP error 404 while getting https://github.com/malnvenshorn/OctoPrint-FilamentManager/archive/1.6.3.zip
ERROR: Could not install requirement https://github.com/malnvenshorn/OctoPrint-FilamentManager/archive/1.6.3.zip because of error 404 Client Error: Not Found for url: https://codeload.github.com/malnvenshorn/OctoPrint-FilamentManager/zip/1.6.3
ERROR: Could not install requirement https://github.com/malnvenshorn/OctoPrint-FilamentManager/archive/1.6.3.zip because of HTTP error 404 Client Error: Not Found for url: https://codeload.github.com/malnvenshorn/OctoPrint-FilamentManager/zip/1.6.3 for URL https://github.com/malnvenshorn/OctoPrint-FilamentManager/archive/1.6.3.zip
Die Aktualisierung wurde nicht erfolgreich abgeschlossen. Bitte konsultiere <code>octoprint.log</code> und <code>plugin_softwareupdate_console.log</code> für Details.

Friendly regards, Robert

advorsky73, 2020-12-14:
same here....
Updating, please wait.
++++++++++++++++++++++++++++++++++++++
Now updating Filament Manager to 1.6.3
++++++++++++++++++++++++++++++++++++++
/home/pi/oprint/bin/python2 -m pip --disable-pip-version-check install https://github.com/malnvenshorn/OctoPrint-FilamentManager/archive/1.6.3.zip --no-cache-dir
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting https://github.com/malnvenshorn/OctoPrint-FilamentManager/archive/1.6.3.zip
HTTP error 404 while getting https://github.com/malnvenshorn/OctoPrint-FilamentManager/archive/1.6.3.zip
Could not install requirement https://github.com/malnvenshorn/OctoPrint-FilamentManager/archive/1.6.3.zip because of error 404 Client Error: Not Found for url: https://codeload.github.com/malnvenshorn/OctoPrint-FilamentManager/zip/1.6.3
Could not install requirement https://github.com/malnvenshorn/OctoPrint-FilamentManager/archive/1.6.3.zip because of HTTP error 404 Client Error: Not Found for url: https://codeload.github.com/malnvenshorn/OctoPrint-FilamentManager/zip/1.6.3 for URL https://github.com/malnvenshorn/OctoPrint-FilamentManager/archive/1.6.3.zip
The update did not finish successfully. Please consult <code>octoprint.log</code> and <code>plugin_softwareupdate_console.log</code> for details.

fefeh1, 2020-12-15:
Same issue here.  I found this in the plugin manager:

Notice regarding plugin "Filament Manager"
Affected versions: <1.6.0
Version 1.6.0 of this plugin is available from a new maintainer, but your version still looks for updates at the repository of the old maintainer. Please re-install the plugin from the repository to update, keeping your settings.

OllisGit, 2021-01-28:
Hi!
as @fefeh1 mentioned, you need to update via the reported URL!

closed 2021-01-28 by OllisGit