OllisGit/OctoPrint-FilamentManager #27: Spool usage not being tracked anymore

ngroulx, 2020-12-05:
Spool usage has stopped being tracked. The amount a spool has is the same both before and after a print.

**Steps to reproduce**
1. Check spool used % and remaining grams
2. Run print
3. After print Check spool remaining again
4. Neither the % used or the remaining g. change at all.

**Expected behavior**
I expect the spool usage % to increase and the remaining g to decrease

**Did the same happen when all other 3rd party plugins are disabled?**
Not sure yet. I've disabled several with no change. 

**Log file**
I had to truncate the log file as it was too large. But you should have the necessary errors in here.
[octoprint.log](https://github.com/OllisGit/OctoPrint-FilamentManager/files/5646495/octoprint.log)

ngroulx, 2021-01-04:
@OllisGit any help here?

OllisGit, 2021-01-28:
Hi,

thx for the logs there are a lot of filament manager errors, please also provide the corresponding gcode, that should help. 

PS. Sorry, for being late to the party, but I made a "Octoprint sabatical" (no coding, no printing) the last months.

ngroulx, 2021-01-28:
No worries @OllisGit, welcome back! Here is the log file and gcode from my current print.

[logs+gcode.zip](https://github.com/OllisGit/OctoPrint-FilamentManager/files/5890178/logs%2Bgcode.zip)

OllisGit, 2021-01-28:
fyi: hmm..looks like there is something wrong in the gcode-parser. I received a warning from OctoPrint itself:
```
GCODE tried to select tool 1250, that looks wrong, ignoring for GCODE analysis
```
GCode line:
```
M204 P1250 R1250 T1250 ; sets acceleration (P, T) and retract acceleration (R), mm/sec^2
```

ngroulx, 2021-01-28:
Not sure what to say, that gcode came straight out of prusaslicer 2.3 and is printing perfectly on my MK3s + mmu2s

Is that affecting the plugin somehow?

OllisGit, 2021-01-29:
plugin-error I need to fix!

OllisGit, 2021-01-29:
Hi @ngroulx ,

new release 1.7.1 is out and should fix some material tracking issue.
After resuming a multi extruder print, the following tracked filament is assigned to tool0. To fix that issue I reimplemented the whole material tracking (maybe it solves other issues as well ;-). 

Please retest with the new version and give me a  feedback!

ngroulx, 2021-01-29:
Awesome! Thanks for the quick fix. I updated to the latest version, and I hope to test it over the weekend.

OllisGit, 2021-01-31:
Hi @ngroulx,

please use Version [1.7.2](https://github.com/OllisGit/OctoPrint-FilamentManager/releases/tag/1.7.2), the previous version is broken!!!

closed 2021-04-19 by OllisGit