OllisGit/OctoPrint-FilamentManager #29: Prusaslicer Beta 2.3.0 'No filament from Meta'

rickyh777, 2020-12-06:
When slicing something with prusaslicer Beta 2.3.0, the filament meta data must have moved as filament Manager no longer detects it. It can therefore no longer determine usage and I get the 'No filament detected from meta' error. I did slice with the fully released prusaslicer version and filament manager worked fine, it has something to do with the beta. Hoping you guys might be able to get out in front of it before the full release and a lot of people start having issues!

Thanks so much for all the development work on this I absolutely love filament manager!

OllisGit, 2021-01-12:
Hi @rickyh777,

sorry to say, but I have no idea what you mean with "No filament detected from meta". I am not the original author of this plugin and in the sourcecode I couldn't find such error message.

Please provide a screenshot and the gcode file, so that I am able to reproduce the issue.

OllisGit, 2021-04-19:
no feedback -> closed

closed 2021-04-19 by OllisGit

git-zombie, 2021-05-30:
I'm slicing from Cura and got the same issue. 

[image]

OllisGit, 2021-06-05:
@git-zombie ticket is closed and related to plugin [CostEstimation-Plugin](https://github.com/OllisGit/OctoPrint-CostEstimation). Solution, see readme (startpage)