OllisGit/OctoPrint-FilamentManager #30: Not getting 2nd color/spool in verification dialog

sg000, 2020-12-07:
Describe the bug
Not getting 2nd color/spool in verification dialog.

Using shared nozzle, 2 extruder setup, both in slicer (PrusaSlicer) and Octoprint setup.
I've tried turning off and on both this plugin and all others.
I DO get the 2 spool selection in the main window.
I DO NOT get 2 spool selection in the verification upon PRINT.
I am certain I used to, even tried an old file that I was sure used to work this way.

Steps to reproduce
Using M600 at color change, but tool is same, T0 because shared nozzle.
PrusaSlicer

Expected behavior
I expect to see 2 spools to select.
WHAT actually tells this plugin that there are 2 different spools in use? Its not clear in the documentation. I was sure the M600 was working despite what the Wiki says.

Did the same happen when all other 3rd party plugins are disabled?
yup

Log file
Dont have a log yet, just turned on logging, but doesn't seem to create a lot just to pop open the dialog.

Screenshots
If applicable, add screenshots to help explain your problem.

Additional context
Add any other context about the problem here.

OllisGit, 2021-01-12:
Hi @sg000 ,

filament change via M600 is not supported for this plugin!
I am planing to implement this feature in the SpoolManager-Plugin.

The gcode includes a T-command which extruder/tool head should be used for extrusion.

Because I don't have a mulit-extruder printer, please provide a gcode file and mention what values do you expect after printing.
Then I will try to reproduce the error in a "virtual printer".

sg000, 2021-01-12:
> filament change via M600 is not supported for this plugin!
> I am planing to implement this feature in the SpoolManager-Plugin.
>
> The gcode includes a T-command which extruder/tool head should be used [...]

That confirms the M600. Documentation seemed to indicate this, but I thought this worked at one point. Perhaps not. No biggie.

I don't have multi either, I do manual changes.

Its not a biggie for me. You can close this out.

closed 2021-01-28 by OllisGit