OllisGit/OctoPrint-FilamentManager #31: Postgres not connect after upgrade octoprint to python 3.7

Hufnagels, 2020-12-18:
Hi there!
I upgraded octoprint to python 3. Until this the database connection to localhost Postgres gone perfectly.
Now the response is: "Failed to connect to the database with the given configuration" in browser, in octoprint log tornado.access - WARNING - 400 POST /plugin/filamentmanager/database/test (fe80::40e:ca3a:4b26:444c) 50.29ms

In the config I didn't changed anything. The DB is ok (checked with psql), Forex. I can connect to it with PSequel without any issue, I see all the data, that is recorded.
Any idea, what's happening?
Thx for help

taker218, 2020-12-18:
Did you install the psycopg2 library in the new Python environment?

https://github.com/OllisGit/OctoPrint-FilamentManager/issues/7#issuecomment-705685436

Hufnagels, 2020-12-18:
Yes, with pip3

Hufnagels, 2020-12-18:
In postman the response is the same: Failed to connect to the database with the given configuration
when I use your api point with the sended json: {"config":{"clientID":"fa5c12bc-40c0-11eb-bee9-98482753d647","name":"octoprint_filamentmanager","password":"*********","uri":"postgresql://localhost","useExternal":true,"user":"******"}}
The credentials in the query used are the same, when I connect via psql. Via psql I can list the tables in the db and the values too.

Hufnagels, 2020-12-19:
Hi, i think is an overall issue too, that octoprint is not optimised to run on p3. On my raspi 2b after upgrade the hole system become slow and unstable (random stops until print, etc). So I decidw to downgradw to 2.7. Your plugin works like a charm. I think, maybe more testing needed on p3.7 with psql connect. 
cheers

taker218, 2020-12-21:
I just had the same problem on a new instance upgraded to python 3.7
I needed to install libpq-dev on my Pi, after that I could connect to the database without any problems.

Daddycool2708, 2020-12-28:
I have the same issue. I can not connect to my db. with the same error tornado.access - WARNING - 400 POST /plugin/filamentmanager/database/test
maybe someone to help me

OllisGit, 2021-01-28:
H @Hufnagels ,
was the hint from @taker218  helpful? Cn the ticket closed?

PS. Sorry, for being late to the party, but I made a "Octoprint sabatical" (no coding, no printing) the last months.

Hufnagels, 2021-01-30:
Nope. Sorry. And came up another issue, wich is opened now: upgrade to 1.7.1:

**2021-01-30 18:48:18,969 - octoprint.plugin.core - ERROR - Error loading plugin filamentmanager
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/plugin/core.py", line 1298, in _import_plugin
    module = _load_module(module_name, spec)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/plugin/core.py", line 69, in _load_module
    return imp.load_module(name, f, filename, details)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_filamentmanager/__init__.py", line 18, in <module>
    from .newodometer import NewFilamentOdometer
ImportError: No module named newodometer**

OllisGit, 2021-04-19:
Solved in FM 1.7.2

closed 2021-04-19 by OllisGit