OllisGit/OctoPrint-FilamentManager #32: Permissions missing/error on older Octoprint

castaway, 2021-01-02:
**Describe the bug**

Filament manager (master as of 2021-01-02), won't start on Octoprint 1.3.12

**Steps to reproduce**
Install Filament Manager
Restart octoprint
Check logs:

**Expected behavior**
Plugin to load and display UI for selection of filament spools etc

**Log file**
```
2021-01-02 15:56:26,313 - octoprint.plugin.core - ERROR - Error loading plugin f
ilamentmanager
Traceback (most recent call last):
  File "/usr/src/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint/plug
in/core.py", line 863, in _import_plugin
    instance = imp.load_module(module_name, f, filename, description)
  File "/usr/src/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint_fila
mentmanager/__init__.py", line 16, in <module>
    from .api import FilamentManagerApi
  File "/usr/src/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint_fila
mentmanager/api/__init__.py", line 19, in <module>
    from octoprint.access.permissions import Permissions
ImportError: No module named access.permissions
```

Is it possible to have a release per Octoprint version that the plugin manager will fetch?

OllisGit, 2021-01-14:
Hi @castaway,

the latest version which works with Python2 and OP1.3 should be Version 0.5.3

Just install this version
https://github.com/OllisGit/OctoPrint-FilamentManager/archive/0.5.3.zip

From technical point of view it is not possible (or I don't know how) to support both permission ways (1.3.x, 1.4.x+), because it is an "Annotation" and not a command in a function.

My suggestion is: update to the latest OP Version 1.5.2...or at least Version 1.4.x

castaway, 2021-01-15:
Thanks ! I'll get around to upgrading eventually, I swear!

closed 2021-01-28 by OllisGit