OllisGit/OctoPrint-FilamentManager #33: filament is not substracted whe option "ask before start" is enabled

diedder, 2021-01-07:
**Describe the bug**
Using a MK3S with MMU2S, plugin MMU Filament select is installed. Filament is not tracked wenn using another tool than 0 an the option "ask bfore start" is enabled. With the option off it works.

**Steps to reproduce**
1. turn option "ask before start on"
2. select different filaments at tool 0 and 4
3. start a print job with only one tool / filament
4. the popup appears and only allows to select the sppol for tool 0 - click ok
5. the popup of the filament selection appears, select filament 5
6. after the print job both filament spool are not substracted
7. turn the option in the settings of
8. repeat steps 2-3
9. the popup of the filament selection appears, select filament 5
10. after the print job the spool on tool 4 is subtracted

**Expected behavior**
always show all 5 tools in the popup, substract the amount on the filament port used. a one color print has in software only one tool (ocotprint shows tool0), but this ony filament can be loaded in another port of the MMU the the first.

**Additional context**
maybe there is a context with #24

OllisGit, 2021-01-28:
Hi @diedder,

could you please provide me a gcode that your are using. That should be helpful.

Is the "MMU2-Thing" using M600 for filament changes? If yes, this plugin doesn't support this gcode command.

OllisGit, 2021-01-29:
Hi @diedder,

to be honest I couldn't reproduce the issue like you described. But regarding the material tracking I fixed a major bug.

 After resuming a multi extruder print, the following tracked filament is assigned to tool0. To fix that issue I reimplemented the whole material tracking (maybe it solves other issues as well ;-). 

Please retest with the new version 1.7.1 and give me a  feedback!

diedder, 2021-01-30:
Hi @OllisGit 
I tried to update to 1.7.1 but it failed. After the update the plugin is not visible anymore. A new install finished with the message "Installed unknown" but the plugin is not visible in the settings... No chance for testing.

Acc. the toolchange. I used the MMU Filament Select Plugin for single color prints with the MMU. So there are no toolchange gcodes necessary. Unfortunately this plugin isn't supported anymore so I have to select the filament directly at the printers display. Multicolor pints can be tested.

OllisGit, 2021-01-31:
Please use the latest version [1.7.2](https://github.com/OllisGit/OctoPrint-FilamentManager/releases/tag/1.7.2) the previous version is broken

Senthiele, 2021-02-07:
1.7.2 is not substracting from the filament manger. I have ran two full spool and nothing has been sub from the spools

diedder, 2021-02-16:
i can confirm this

HappyWarbird, 2021-03-06:
I can confirm this too. With 1.7.2 and the "ask before start" option activated, filament usage isn't substracted from the active spool. I have only 1 Extruder on my snapmaker original.

OllisGit, 2021-06-05:
Please provide the octoprint log file. The should include these lines:
```
2021-06-05 14:18:45,909 - octoprint.plugins.filamentmanager - DEBUG - Printer started
2021-06-05 14:18:45,909 - octoprint.plugins.filamentmanager - DEBUG - Odometer: On
2021-06-05 14:18:45,909 - octoprint.plugins.filamentmanager - DEBUG - AutoPause: Off
2021-06-05 14:19:49,263 - octoprint.plugins.filamentmanager - INFO - Updating Filament usage for octoprint configured toolcount: 5
2021-06-05 14:19:49,264 - octoprint.plugins.filamentmanager - INFO - Filament tracked toolcount: 1
2021-06-05 14:19:49,264 - octoprint.plugins.filamentmanager - INFO - Filament tracked values: [210.82672000000017]
2021-06-05 14:19:49,264 - octoprint.plugins.filamentmanager - INFO - Filament used: 210.82672000000017 mm (tool0)
2021-06-05 14:19:49,270 - octoprint.plugins.filamentmanager - INFO - Updated remaining filament on spool 'Number2 - PLA (MatterMost)' from 1939.9399804807058g to 1934.6180419849065g (-5.321938495799259g)

```

diedder, 2021-06-06:
sorry cant't help, I switched to SpoolManagerPlugin since multi tool is supported.