OllisGit/OctoPrint-FilamentManager #34: Hide empty Spools in Sidebar

realbadfox, 2021-01-20:
Hallo,

really great Plugin, but is it possible to hide empty Spools in the Sidebar in Octoprint, so that the list won't be so long?
You can make a checkbox in the settings of the Plugin to enable or disable that feature.

yours Chris

**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

OllisGit, 2021-01-28:
Hi,

this request is duplicated, see #6!
PS. it is already implemented, see https://github.com/OllisGit/OctoPrint-SpoolManager/issues/81

closed 2021-01-28 by OllisGit

realbadfox, 2021-01-31:
But i would love to have it in the filament manager not spool manager