OllisGit/OctoPrint-FilamentManager #35: Inable to conenct to external database.

blindguy, 2021-01-24:
Describe the bug
Trying to connect the filament manager to a external database and I keep getting this message from three different octoprint pi's.
the 192.168.1.1 is my local PC IP address. The postgresql DB is sitting on my server and I can access it via DBever.
WARNING - 400 POST /plugin/filamentmanager/database/test (::ffff:192.168.1.1) 21.05ms
they docs say to run this:
~/OctoPrint/venv/bin/pip install psycopg2 but that file/direcotry does not exist for me:
root@octopi03:/home/pi# ~/OctoPrint/venv/bin/pip install psycopg2
bash: /root/OctoPrint/venv/bin/pip: No such file or directory

Steps to reproduce

run this:
~/OctoPrint/venv/bin/pip install psycopg2
Expected behavior
I expect the plugin to connect to the DB

Did the same happen when all other 3rd party plugins are disabled?
Everything else works fine

blindguy, 2021-01-24:
Everyone say to use the PIP? but that does not work.   any thoughts?

pip install psycopg2
bash: pip: command not found

blindguy, 2021-01-24:
Some other people say to use ver 3?
pip3 psycopg2 

I cant run that one either.

blindguy, 2021-01-24:
The notes also say to use pacman to install postgresql-libs if problems
sudo pacman -S postgresql-libs

I did a apt install pacman and it said it installed somethign but the command still does not work.
sudo pacman -S postgresql-libs
sudo: pacman: command not found

blindguy, 2021-01-24:
tried doing oprint/bin/pip install psycopg2
and got lots of errorss.

Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting psycopg2
  Using cached psycopg2-2.8.6.tar.gz (383 kB)
    ERROR: Command errored out with exit status 1:
     command: /home/pi/oprint/bin/python2 -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-jm4iyc/psycopg2/setup.py'"'"'; __file__='"'"'/tmp/pip-install-jm4iyc/psycopg2/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' egg_info --egg-base /tmp/pip-pip-egg-info-ouEujv
         cwd: /tmp/pip-install-jm4iyc/psycopg2/
    Complete output (23 lines):
    running egg_info
    creating /tmp/pip-pip-egg-info-ouEujv/psycopg2.egg-info
    writing /tmp/pip-pip-egg-info-ouEujv/psycopg2.egg-info/PKG-INFO
    writing top-level names to /tmp/pip-pip-egg-info-ouEujv/psycopg2.egg-info/top_level.txt
    writing dependency_links to /tmp/pip-pip-egg-info-ouEujv/psycopg2.egg-info/dependency_links.txt
    writing manifest file '/tmp/pip-pip-egg-info-ouEujv/psycopg2.egg-info/SOURCES.txt'

    Error: pg_config executable not found.

    pg_config is required to build psycopg2 from source.  Please add the directory
    containing pg_config to the $PATH or specify the full executable path with the
    option:

        python setup.py build_ext --pg-config /path/to/pg_config build ...

    or with the pg_config option in 'setup.cfg'.

    If you prefer to avoid building psycopg2 from source, please install the PyPI
    'psycopg2-binary' package instead.

    For further information please check the 'doc/src/install.rst' file (also at
    <https://www.psycopg.org/docs/install.html>).

    ----------------------------------------
ERROR: Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.
p

OllisGit, 2021-01-28:
Hi @blindguy,

did you follow these instructions?

https://github.com/OllisGit/OctoPrint-FilamentManager/wiki/Setup-PostgreSQL-on-Raspbian-(Stretch)

After DB-Installation follow this chapter: https://github.com/OllisGit/OctoPrint-FilamentManager/wiki/Setup-PostgreSQL-on-Raspbian-(Stretch)#configure-filamentmanager

blindguy, 2021-01-28:
@OllisGit Thanks for your attention to this matter.

As I may reimage and reinstall any of my PI's at any time I installed the PostgreSQL DB on a actual servers that I have,
I have configure it for access to my internal network I can connect to that PostgreSQL from other servers such as my windows box using DBever and other apps.

When I do the 
$ sudo apt-get update
$ sudo apt-get install python-dev libpq-dev

I see this:

libpq-dev is already the newest version (11.9-0+deb10u1).
python-dev is already the newest version (2.7.16-1).

The log never indicates anything except for:

2021-01-28 18:23:07,178 - tornado.access - WARNING - 400 POST /plugin/filamentmanager/database/test (::ffff:192.168.1.1) 46.61ms
2021-01-28 18:23:07,358 - tornado.access - WARNING - 400 POST /plugin/filamentmanager/database/test (::ffff:192.168.1.1) 46.24ms
2021-01-28 18:23:07,521 - tornado.access - WARNING - 400 POST /plugin/filamentmanager/database/test (::ffff:192.168.1.1) 53.26ms
2021-01-28 18:23:07,654 - tornado.access - WARNING - 400 POST /plugin/filamentmanager/database/test (::ffff:192.168.1.1) 40.84ms
2021-01-28 18:23:07,814 - tornado.access - WARNING - 400 POST /plugin/filamentmanager/database/test (::ffff:192.168.1.1) 42.30ms
2021-01-28 18:23:08,196 - tornado.access - WARNING - 400 POST /plugin/filamentmanager/database/test (::ffff:192.168.1.1) 267.95ms
2021-01-28 18:23:08,295 - tornado.access - WARNING - 400 POST /plugin/filamentmanager/database/test (::ffff:192.168.1.1) 45.99ms

blindguy, 2021-01-28:
@OllisGit I'm folling your steps to see if I can get it working form one of the PI's.

Thorondor2011, 2021-02-13:
I had the same problem, the plugin not being able to connect to external database. I fixed it by installing the Postgresql client software with

sudo apt-get install postgresql-client

Hope this helps people who are having troubles.

blindguy, 2021-02-14:
@Thorondor2011 thanks for the info but that didn't help me much.  

2021-02-13 17:56:27,348 - tornado.access - ERROR - 500 POST /plugin/filamentmanager/database/test (::ffff:192.168.1.1) 174.13ms
2021-02-13 17:56:27,927 - octoprint.plugins.filamentmanager - WARNING - self.filamentManager is not initialized yet
2021-02-13 17:56:27,928 - octoprint - ERROR - Exception on /plugin/filamentmanager/database/test [POST]
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/flask/app.py", line 2447, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/flask/app.py", line 1953, in full_dispatch_request
    return self.finalize_request(rv)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/flask/app.py", line 1968, in finalize_request
    response = self.make_response(rv)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/flask/app.py", line 2098, in make_response
    "The view function did not return a valid response. The"
TypeError: The view function did not return a valid response. The function either returned None or ended without a return statement.
2021-02-13 17:56:28,087 - tornado.access - ERROR - 500 POST /plugin/filamentmanager/database/test (::ffff:192.168.1.1) 171.84ms

Thorondor2011, 2021-02-14:
I'm not sure, but if I read your comments correctly you still haven't installed psycopg2. The command I used to install psycopg2 was:

/home/pi/oprint/bin/python -m pip install psycopg2

Found the pip install command syntax on Google.
Here's hoping it helps.

blindguy, 2021-02-14:
@Thorondor2011 thanks for all the help.  I was able to reformat and reinstall octoprint from scratch and now have it running on that pi and I can connect to the from there. When tryignt to connect from my two other octoprints I still get stuff like this:

PI 1:
2021-02-14 10:15:19,362 - octoprint - ERROR - Exception on /plugin/filamentmanager/database/test [POST]
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/flask/app.py", line 2447, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/flask/app.py", line 1953, in full_dispatch_request
    return self.finalize_request(rv)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/flask/app.py", line 1968, in finalize_request
    response = self.make_response(rv)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/flask/app.py", line 2098, in make_response
    "The view function did not return a valid response. The"
TypeError: The view function did not return a valid response. The function either returned None or ended without a return statement.
2021-02-14 10:15:19,559 - tornado.access - ERROR - 500 POST /plugin/filamentmanager/database/test (::ffff:192.168.1.1) 224.60ms

P2:
2021-02-14 17:14:02,449 - octoprint - ERROR - Exception on /plugin/filamentmanager/database/test [POST]
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/flask/app.py", line 2447, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/flask/app.py", line 1953, in full_dispatch_request
    return self.finalize_request(rv)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/flask/app.py", line 1968, in finalize_request
    response = self.make_response(rv)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/flask/app.py", line 2098, in make_response
    "The view function did not return a valid response. The"
TypeError: The view function did not return a valid response. The function either returned None or ended without a return statement.
2021-02-14 17:14:02,472 - tornado.access - ERROR - 500 POST /plugin/filamentmanager/database/test (::ffff:192.168.1.1) 44.44ms

/home/pi/oprint/bin/python -m pip install psycopg2
it says its already installed.

pi@octopi03:~ $ /home/pi/oprint/bin/python -m pip install psycopg2
DEPRECATION: Python 2.7 reached the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 is no longer maintained. pip 21.0 will drop support for Python 2.7 in January 2021. More details about Python 2 support in pip can be found at https://pip.pypa.io/en/latest/development/release-process/#python-2-support pip 21.0 will remove support for this functionality.
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: psycopg2 in ./oprint/lib/python2.7/site-packages (2.8.6)

apprecaite all help

blindguy, 2021-02-14:
@Thorondor2011 Success!!

I went in and deleted the filament manager on the other two boxes and then reinstalled and I can now hit the db on one of the boxes.

It's not the best solution but I think I got it going!!! Thanks SO MUCH!!!

Thorondor2011, 2021-02-14:
Glad you got it to work.

closed 2021-04-11 by OllisGit