OllisGit/OctoPrint-FilamentManager #38: Update via Ocotprint PluginManage Update failed 1.7.0 -> 1.7.1

LingChristian, 2021-01-29:
Hey there,

I tryed to update FailamentManage via Ocotoprint -> Software Update  ... It says evrything updated fine, but the Control Panel on the left bottom for the Filament is now gone. Also the plugin disapperead from the Pluginmanager. I can try to reinstall (1.7.0.) ... he´s installing it without any error but also isn´t working in the end.

Any ideas?

Shama71, 2021-01-29:
Same problem
env.python.version: 3.7.3
octoprint.version: 1.5.3

moong8te, 2021-01-29:
Same here. Update seemed to be running fine. Octoprint restarted. First message that appeared was this:
[image]

In the configuration the plugin is missing as well as in the plug-in manager under installed plugins.
I then tried to re-install Filament Manager 1.7.1 from the Plugin Manager.
As you can see it found an existing 1.7.1 version but with the issues described above.
Also this installation completed successfully as far as the status window is concerned.
[image]

But after the reboot the same error of missing Filament Manager.

I am also on 
env.python.version: 3.7.3
octoprint.version: 1.5.3
just like Shama71

patofoto, 2021-01-30:
Having same issue. After update plugin disappeared. Cannot re-install.

ptsiampas, 2021-01-30:
Same, I just did the update.  Hope you can resolve it soon, not a deal-breaker but hey we do love the plugin.

mikepaxton, 2021-01-30:
Same issue.  Just setup a new Raspberry Pi with Octoprint 1.5.3 on it and installed Filament Manager.  Noticed after install the pop up notification said "Unkown" install.  After reboot filament manager is not installed or at least not showing in Plug-in Manager.

Noirvent, 2021-01-30:
Can confirm, same issue. Update work without giving error, but the plugin is nowhere to be found on octopi.
Tried the install from pluginmanager and manually with pip.

JohnVodopija, 2021-01-30:
Same issue. Chose to update Filament Manager from 1.7.0 to 1.7.1. The onscreen install progress indicated successful upgrade. FilamentManager is no longer visible in list of installed plugins. Tried to reinstall plugin from PluginManager with same result. Help!

OctoPrint 1.5.3 Python 2.7.16 OctoPi 0.17.0

gx1400, 2021-01-30:
Same as above, upgraded from 1.7.0 to 1.7.1. said it was a successful update but disappeared from Plugin Manager. Cleanup tab shows filamentmanager available for cleanup

yohannaftali, 2021-01-30:
same, found this in octoprint.log, seems 1.7.1 requure NewFilamentOdometer module that not installed

2021-01-30 03:36:13,493 - octoprint.plugin.core - ERROR - Error loading plugin filamentmanager
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/plugin/core.py", line 1298, in _import_plugin
    module = _load_module(module_name, spec)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/plugin/core.py", line 69, in _load_module
    return imp.load_module(name, f, filename, details)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_filamentmanager/__init__.py", line 18, in <module>
    from .newodometer import NewFilamentOdometer
ImportError: No module named newodometer
2021-01-30 03:36:19,467 - octoprint.plugin.core - INFO - Plugin Anonymous Usage Tracking is disabled.
2021-01-30 03:36:20,120 - octoprint.plugin.core - ERROR - Error loading plugin filamentmanager
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/plugin/core.py", line 1298, in _import_plugin
    module = _load_module(module_name, spec)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/plugin/core.py", line 69, in _load_module
    return imp.load_module(name, f, filename, details)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_filamentmanager/__init__.py", line 18, in <module>
    from .newodometer import NewFilamentOdometer
ImportError: No module named newodometer
2021-01-30 03:36:20,552 - octoprint.plugins.pluginmanager - WARNING - The plugin was installed successfully, but couldn't be found afterwards to initialize properly during runtime. Please restart OctoPrint.
2021-01-30 03:36:20,555 - octoprint.util - WARNING - Could not delete temporary directory /tmp/tmpIE5xQj: [Errno 21] Is a directory: '/tmp/tmpIE5xQj'

MakeItSlow, 2021-01-30:
Same here. Found this in the plugin manager log:

2021-01-29 22:16:39,900 > Building wheels for collected packages: OctoPrint-FilamentManager
2021-01-29 22:16:39,900 > Building wheel for OctoPrint-FilamentManager (setup.py): started
2021-01-29 22:16:44,548 > Building wheel for OctoPrint-FilamentManager (setup.py): finished with status 'done'
2021-01-29 22:16:44,548 > Created wheel for OctoPrint-FilamentManager: filename=OctoPrint_FilamentManager-1.7.1-cp27-none-any.whl size=62853 sha256=250860b2d1db18f68b673f4bb6cde876eff632dd013aa55e99500d67e054c499
2021-01-29 22:16:44,549 > Stored in directory: /tmp/pip-ephem-wheel-cache-cTzH2V/wheels/36/c7/bb/b6f4896b98000f8a43853d43e7ff6d0d2e9151c94e09b1a0f3
2021-01-29 22:16:44,550 > Successfully built OctoPrint-FilamentManager
2021-01-29 22:16:44,550 > Installing collected packages: OctoPrint-FilamentManager
2021-01-29 22:16:44,550 > Found existing installation: OctoPrint-FilamentManager 1.7.1
2021-01-29 22:16:44,551 > Uninstalling OctoPrint-FilamentManager-1.7.1:
2021-01-29 22:16:44,551 > Successfully uninstalled OctoPrint-FilamentManager-1.7.1
**2021-01-29 22:16:44,551 > Successfully installed OctoPrint-FilamentManager-1.7.1**

Yet it's not in Plugin Manager and not in list of installed plugins

Seqway, 2021-01-30:
For me the same !

gx1400, 2021-01-30:
https://github.com/OllisGit/OctoPrint-FilamentManager/commit/5237c4dc48f9eba0f26d50c042a938649ecc7bf9
New commit shows:
(-) from .odometer import FilamentOdometer
(+) from .newodometer import NewFilamentOdometer

jives with the following lines from octoprint.log:

2021-01-29 20:39:51,894 - octoprint.plugin.core - ERROR - Error loading plugin filamentmanager
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/plugin/core.py", line 1298, in _import_plugin
    module = _load_module(module_name, spec)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/plugin/core.py", line 69, in _load_module
    return imp.load_module(name, f, filename, details)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/vendor/imp.py", line 238, in load_module
    return load_package(name, filename)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/vendor/imp.py", line 212, in load_package
    return _load(spec)
  File "<frozen importlib._bootstrap>", line 696, in _load
  File "<frozen importlib._bootstrap>", line 677, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 728, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_filamentmanager/__init__.py", line 18, in <module>
    from .newodometer import NewFilamentOdometer
ModuleNotFoundError: No module named 'octoprint_filamentmanager.newodometer'

I'm sure it'll be a quick fix, looks like maybe a new filament odometer wasn't added?

JHale716, 2021-01-30:
Confirming the same error on update from 1.7.0 to 1.7.1. Rinse and repeat of the logs.

dgriswo, 2021-01-30:
You can re-install version 1.7.0 using the Plugin Manager.  Go to the "Get More" link in the Plugin Manager and paste the 1.7.0 release url: https://github.com/OllisGit/OctoPrint-FilamentManager/releases/download/1.7.0/master.zip

yohannaftali, 2021-01-30:
> You can re-install version 1.7.0 using the Plugin Manager. Go to the "Get More" link in the Plugin Manager and paste the 1.7.0 release url: https://github.com/OllisGit/OctoPrint-FilamentManager/releases/download/1.7.0/master.zip

yes, i already done that, but now problem with odometer seems wrongly calculate weight

domhardt, 2021-01-30:
Same problem for me. Manual install of version 1.7.0 fixed the problem for now.

AndrewStick, 2021-01-30:
Same problem. Downgraded to 1.7.0

Cyberbastler, 2021-01-30:
Same problem

nokemono42, 2021-01-30:
Same problem. octoprint.log says:

`
2021-01-30 12:57:12,340 - octoprint.plugin.core - INFO - Loading plugins from /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins, /root/.octoprint/plugins and installed plugin packages...

2021-01-30 12:57:15,764 - octoprint.plugin.core - ERROR - Error loading plugin filamentmanager
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/plugin/core.py", line 1298, in _import_plugin
    module = _load_module(module_name, spec)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/plugin/core.py", line 69, in _load_module
    return imp.load_module(name, f, filename, details)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_filamentmanager/__init__.py", line 18, in <module>
    from .newodometer import NewFilamentOdometer
`

realbadfox, 2021-01-30:
Same Problem here.

Hipsta69, 2021-01-30:
Same issue.

After trying to reinstall I get a message saying done but popup says installed: unknown
[image]

Jack77777777, 2021-01-30:
Same issue here.

browser.user_agent: Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/88.0.4324.104 Safari/537.36
connectivity.connection_check: 8.8.8.8:53
connectivity.connection_ok: true
connectivity.enabled: true
connectivity.online: true
connectivity.resolution_check: octoprint.org
connectivity.resolution_ok: true
env.hardware.cores: 4
env.hardware.freq: 1500
env.hardware.ram: 3959984128
env.os.bits: 32
env.os.id: linux
env.os.platform: linux
env.plugins.pi_support.model: Raspberry Pi 4 Model B Rev 1.1
env.plugins.pi_support.octopi_version: 0.18.0
env.plugins.pi_support.throttle_state: 0x0
env.python.pip: 20.3.3
env.python.version: 3.7.3
env.python.virtualenv: true
octoprint.safe_mode: false
octoprint.version: 1.5.3
printer.firmware: Marlin 1.1.9 (Github, davidramiro) MARLIN-

mpd352, 2021-01-30:
Same issue.

Jack77777777, 2021-01-30:
Downgraded to 0.1.7. Database before still working and the data before upgrade is still there. Luckily i only printed about 10g when i saw Filamentmanager diappeared.

bakfiets2, 2021-01-30:
Same here. Downgraded

shogun168, 2021-01-30:
Same here .. I hope I didn't lose filament history, I was in the middle of recording 20 ish rolls

JohnVodopija, 2021-01-30:
@shotgun168 you won’t lose your filament data. I followed the instructions
in this thread to downgrade to 1.7.0 and my filament data was presented and
intact.

daclarkson, 2021-01-30:
Same problems and log messages.  I tried a few updates and after they all failed I downgraded.  Lost all data though.

moong8te, 2021-01-30:
> Same problems and log messages. I tried a few updates and after they all failed I downgraded. Lost all data though.

Sorry to hear that your data seems to be lost.

I downgraded by going to

settings -> plugin manager -> scroll down to "get more" -> scroll down to "... from URL" and pasted the following link to install the former version:

https://github.com/OllisGit/OctoPrint-FilamentManager/releases/download/1.7.0/master.zip

In my case all data remained using this procedure.

daclarkson, 2021-01-31:
> Sorry to hear that your data seems to be lost.
>
> I downgraded by going to
>
> settings -> plugin manager -> scroll down to "get more" -> scroll down to "... from URL" and pasted the following link to [...]

I've downgraded the same way.  I rebooted the Pi a few times between upgrade attempts so I may of overwritten/corrupted something during the process.

djlightflighter, 2021-01-31:
same issue it went missing after the install and my OCTO Dash is freaking out about it

Turtletrumpet, 2021-01-31:
The downgrade works for me, too. Thanks!

Moskito99, 2021-01-31:
Same Problem!
```
2021-01-31 08:15:23,797 - octoprint.plugin.core - ERROR - Error loading plugin filamentmanager
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/plugin/core.py", line 1298, in _import_plugin
    module = _load_module(module_name, spec)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/plugin/core.py", line 69, in _load_module
    return imp.load_module(name, f, filename, details)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/vendor/imp.py", line 238, in load_module
    return load_package(name, filename)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/vendor/imp.py", line 212, in load_package
    return _load(spec)
  File "<frozen importlib._bootstrap>", line 696, in _load
  File "<frozen importlib._bootstrap>", line 677, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 728, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_filamentmanager/__init__.py", line 18, in <module>
    from .newodometer import NewFilamentOdometer
ModuleNotFoundError: No module named 'octoprint_filamentmanager.newodometer'
```

OllisGit, 2021-01-31:
SORRY,
missed to add the new filament tracker python file to git repository.

It is added in new Version 1.7.2!

BR
Olli

daclarkson, 2021-01-31:
> SORRY,
> missed to add the new filament tracker python file to git repository.
> 
> It is added in new Version 1.7.2!
> 
> BR
> Olli

Is there a way to query files/database directly to try and recover data?

OllisGit, 2021-01-31:
Just for the records: Version 1.7.1 didn't crash the database all data is still there!

The database file is located here:
```
<op-installation>/data/filamentmanager/filament.db
```
There is no backup mechanism or direct download.

[Advertisement] If you wan't this functionality, use [SpoolManager](https://github.com/OllisGit/OctoPrint-SpoolManager). Every plugin-installations starts a backup of the database.

daclarkson, 2021-01-31:
> Just for the records: Version 1.7.1 didn't crash the database all data is still there!
>
> The database file is located here:
>
> ```
> <op-installation>/data/filamentmanager/filament.db
> ```
>
> There is no [...]

Thanks for the pointer to the correct location.

For clarity this crash/corruption is clearly my fault and not this package.  I was a little too impatient and it led to a overwrite of the db file.  I've had a quick look with `sqlite3` and my 32K file has database tables in it but no data.

Every day's a school day.  :)

yohannaftali, 2021-01-31:
just upgrade to 1.7.2, new problem display layer progress become missing? is it correlated or separated problem?

2021-01-31 08:45:14,514 - octoprint.plugin.core - WARNING - Plugin LayerDisplay (0.4.3) is not compatible to Python 3.7.3 (compatibility string: >=2.7,<3).
2021-01-31 08:45:14,576 - octoprint.plugin.core - ERROR - Error loading plugin DisplayLayerProgress
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/plugin/core.py", line 1298, in _import_plugin
    module = _load_module(module_name, spec)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/plugin/core.py", line 69, in _load_module
    return imp.load_module(name, f, filename, details)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/vendor/imp.py", line 238, in load_module
    return load_package(name, filename)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/vendor/imp.py", line 212, in load_package
    return _load(spec)
  File "<frozen importlib._bootstrap>", line 696, in _load
  File "<frozen importlib._bootstrap>", line 677, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 728, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_DisplayLayerProgress/__init__.py", line 27, in <module>
    from octoprint_DisplayLayerProgress.CachedSettings import CachedSettings
ModuleNotFoundError: No module named 'octoprint_DisplayLayerProgress.CachedSettings'
2021-01-31 08:45:14,766 - octoprint.plugins.pluginmanager - WARNING - The plugin was installed successfully, but couldn't be found afterwards to initialize properly during runtime. Please restart OctoPrint.

djlightflighter, 2021-01-31:
I noticed that in 1.71 also the Layer Properties Plugin disappears when you install 1.71 and 1.72 now also?

OllisGit, 2021-01-31:
Hi @djlightflighter,

what is the "Layer Properties Plugin"? Do you mean "DisplayLayerProgress"?
If yes, please install the latest version [1.25.3](https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/releases/tag/1.25.3). That solved the mentioned error above.

djlightflighter, 2021-01-31:
Thanks brother enjoy that coffee !!!!!

JHale716, 2021-02-06:
Have updated from 1.7.0 to 1.7.2 today without any obvious issues. Yet to test further than install.

Jack77777777, 2021-04-13:
@OllisGit is Filamentmanager 1.7.2 working with PrintJobHistory V1.11.1-dev. You mentioned 1.7.2 is not working with PrintJobHistory <= V1.10.0. But i can´t find if it is fixed with PrintJobHistory V1.11.1-dev. I need to use Filamentmanager until Spool Manager is supported by Octodash.

closed 2021-04-19 by OllisGit