OllisGit/OctoPrint-FilamentManager #40: PrintJobHistory 1.10.0 not working with FilamentManager1.7.2

Moskito99, 2021-01-31:
**Describe the bug**
The Plugin PrintJobHistory doesn't work with FilamentManager 1.7.2

**Log file**
```
021-01-31 15:07:34,467 - octoprint.plugin - ERROR - Error while calling plugin PrintJobHistory
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/plugin/__init__.py", line 271, in call_plugin
    result = getattr(plugin, method)(*args, **kwargs)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/util/__init__.py", line 1890, in wrapper
    return f(*args, **kwargs)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_PrintJobHistory/__init__.py", line 620, in on_event
    self._printJobFinished("canceled", payload)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_PrintJobHistory/__init__.py", line 436, in _printJobFinished
    self._createAndAssignFilamentModel(self._currentPrintJobModel, payload)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_PrintJobHistory/__init__.py", line 254, in _createAndAssignFilamentModel
    filemanentModel.usedLength = self._filamentManagerPluginImplementation.filamentOdometer.totalExtrusion[0]
AttributeError: 'FilamentManagerPlugin' object has no attribute 'filamentOdometer'
```

rufinus, 2021-01-31:
see https://github.com/OllisGit/OctoPrint-FilamentManager/releases/tag/1.7.2

Moskito99, 2021-01-31:
OK, thanks!
Going back to 1.7.0, because without PrintJobHistory it makes no sense.

closed 2021-01-31 by Moskito99