OllisGit/OctoPrint-FilamentManager #41: Hide Empty spools shows spools that is more than 100% used (inactive spools)

realbadfox, 2021-02-01:
Hallo,

is it possible to hide empty spools shows spools that is more than 100% used (inactive spools)?
Would be really great if the Filament Manager can do this same like SpoolManager

Could it be possible to manually mark a spool as empty, instead of correcting the usage amount?

yours chris

cmoe71, 2021-03-21:
I would like to have that implemented too as I don't like to delete empty spools for history reasons!
(Thanks for maintaining this plugin)

OllisGit, 2021-04-19:
@realbadfox please delete the PullRequest in the old Git-Hub repository....or did you already implement this feature.
https://github.com/malnvenshorn/OctoPrint-FilamentManager/pull/234
Thx, in advance
Olli

OllisGit, 2021-04-20:
Hi,
something like that:
[image]

cmoe71, 2021-04-20:
Would be o.k. for me, best regards, Clemens

OllisGit, 2021-05-24:
Hi @realbadfox, @cmoe71,
latest version 1.8.0 include this feature.
Please test and give feedback.

Thx, in advance
Olli

cmoe71, 2021-05-24:
Hi Olli,

played around with it - works fine for me, thanks for your efforts!

Best regards,
Clemens

closed 2021-05-25 by OllisGit