OllisGit/OctoPrint-FilamentManager #43: Request of some functions

FlexerJR, 2021-02-02:
Hello OllisGit,

thanks for maintaining this plugin. I have some improvements for this one. The biggest thing is, that the way to the plugin for the user is horrible. I have to click on settings and then i have to scroll down to the endless list of plugins to find Filament Manager. 

It would be awesome to have one button to pop up the Filament Manager just with ONE click. It is so annoying. Sorry but somebody had to sayit.

Some more improvments:
-Please make it possible to change spools during the print. I'm changing spool due to different print colors. 
-Make it possible to edit all passive spools during the print. (Not for the active spool during printing)
-I use Spoolmanager as well. There are a lot more options to select. Just take a look at the image 

[image]

Best regards

locki-cz, 2021-03-27:
+ add some picture ... spool or colour

OllisGit, 2021-04-19:
Hi @FlexerJR,

your first point was the reason why I started with SpoolManager ;-) 

Changing Spools is currently not supported in SpoolManager as well, but it is on the roadmap.

@locki-cz these two feature are already implemented in SpoolManager-Plugin.

closed 2021-06-05 by OllisGit