OllisGit/OctoPrint-FilamentManager #46: Plugin installs but is not seen my octoprint (install unknown)

ToxicCrystal2, 2021-02-27:
I had downloaded this plugin before a couple weeks ago and it worked fine, but now I'm having problems.

I went through the plugin manager and clicked install. when it finished the window popped up but said this

```
Plugin management log
Install Unknown: 
A restart is needed for the changes to take effect.
```
when I restart its not visible anywhere.

I haven't run into any other plugins doing the same thing, i tried installing the developer version and it does install and show in the plugin manager but its not usable. (I mainly used that so i could uninstall the whole thing because im not well versed in the command line areas like CMD or PuTTY)

**Log file**
[octoprint.log](https://github.com/OllisGit/OctoPrint-FilamentManager/files/6055607/octoprint.log)

[plugin_pluginmanager_console.log](https://github.com/OllisGit/OctoPrint-FilamentManager/files/6055622/plugin_pluginmanager_console.log)
(also included incase necessary, I did reinstall the plugin multiple times without uninstalling cause it didn't show up in octoprint for me to click uninstall. Hence installing the developer version.)

**Screenshots**

[image]

Octoprint 1.5.3
Win10 / Chrome
Pi 2B Rev 1.1

lowellPrime, 2021-04-04:
@ToxicCrystal2 
There's a good chance this is may be the same issue as https://github.com/OllisGit/OctoPrint-FilamentManager/issues/23, as the behavior you're describing here is exactly what you see with that underlying issue.  If you check the Octoprint Server output, do you see the following?

  File "c:\octoprint\venv\lib\site-packages\sqlalchemy\util\compat.py", line 172, in <module>
    time_func = time.clock
AttributeError: module 'time' has no attribute 'clock'
2021-04-04 08:08:49,405 - octolapse.__init__ - INFO - Release mode detected.
2021-04-04 08:08:49,436 - octoprint.plugins.pluginmanager - WARNING - The plugin was installed successfully, but couldn't be found afterwards to initialize properly during runtime. Please restart OctoPrint.

OllisGit, 2021-04-19:
closed, because duplicated. See #23

closed 2021-04-19 by OllisGit