OllisGit/OctoPrint-FilamentManager #48: Varied currency options

NandoGommez, 2021-03-02:
**Is your feature request related to a problem? Please describe.**
It's not exactly a problem, but it frustrates me not to be able to choose the currency of the filament cost. The only icon and currency, is in euros: €

**Describe the solution you'd like**
Well, the solution I suggest, is to put a drop-down menu or that famous drop-down select. (same when choosing the vendor profile) with currency alternatives, such as "USD, BRL"

**Describe alternatives you've considered**
I had no alternative to this "problem"

**Additional context**
Well, that's it, it's not necessarily a problem but I believe that several people like me, would be happy to choose our currency of origin for calculations, even if they are not accurate calculations, but the Plugin helps a lot.
By the way, a great plugin, I'm using it as a bank to register all my filaments.

Before I used CURA, however it does not show how much filament was used in real time printing.

A great job to all the devs involved.

AndyPC54, 2021-03-02:
There is an option to set the Currency Symbol in the Filament Manager, Plug-in Configuration, Appearance tab.  This is under the 'spanner' icon top right of the Filament Spools page.

I am using FM version 1.6.3 and can set £ (UK Pound) instead of €.

Regards,
Andy

NandoGommez, 2021-03-03:
Oh, I hadn't seen it, thanks for the tip.

closed 2021-03-03 by NandoGommez