OllisGit/OctoPrint-FilamentManager #50: Nozzle heat to high

BenDavidson90, 2021-03-15:
Hi,

Since some days, I noticed thant my printer heat the nozzle 60° higher than what was set in Cura.
I tested first by printing my files directly on the printer and all is OK so the problem is relative to octoprint.
I came across [this issue](https://github.com/malnvenshorn/OctoPrint-FilamentManager/issues/235) on the source repo that is the same as me.
So, I disabled Filament Manager, tested the same files as before and  the problem disappear, so I concluded that the problem is finaly comming from the Filament Manager plugin.

Is the problem known ?

OllisGit, 2021-04-19:
Hi @BenDavidson90,
not known!

Did you already checked your spool configuration in FilamentManager plugin?

In this example the selected spool has on offset of ```-20``` and in gcode ```M104 S200```. The result after print was started looks like this:

[image]

BenDavidson90, 2021-04-28:
Hi,
I did few tests today and it turns out that it was my fault: I had set the temperature offset of the spool at 200...
I use SpoolManager too and as in this one you can adjust the filament temperature I must have confused the two. Sorry !

closed 2021-04-28 by BenDavidson90