OllisGit/OctoPrint-FilamentManager #51: Korrektur Fluss pro Rolle

hawkeyexp, 2021-03-16:
Es wäre super wenn man im Filament-Manager auch den Fluss beeinflussen könnte da es immer mal Abweichungen pro Hersteller gibt und da ich OctoDash verwende welches dieses Plugin gut unterstützt wäre es ein tolles Feature - der SpoolManaer hat es ja soweit ich gesehen habe aber dieser wird wiedrum nicht von OctoDash unterstützt.

OllisGit, 2021-10-16:
Hi @hawkeyexp,

I think there is a Beta-Version of OctoDash 2.3, see https://github.com/UnchartedBull/OctoDash/issues/676#issuecomment-896807800

Adding a new database-field took a lot of time, so from my perspective I will invest this time to implement the external database feature in SpoolManager.

hifihedgehog, 2022-01-07:
@UnchartedBull, I didn't happen to see any beta and the nightly link on the Readme is also outdated. Any ideas when we might get SpoolManager support?

UnchartedBull, 2022-01-17:
You can download a dev build from OctoDash here: https://github.com/jneilliii/OctoDash/actions/runs/1527977960 (just click on download artifact). Final release should hopefully be out soon, since I have a litte more time now.