OllisGit/OctoPrint-FilamentManager #53: Can't connect to database

tonnic, 2021-03-27:
**Describe the bug**
I have 3 octoprint servers and an external postgres server.  2 of the 3 can connect perfectly, but one of them cannot.

**Steps to reproduce**
Added the credentials and clicked test and stays red... other 2 are perfectly fine.

**Expected behavior**
Green test button and connection to database.

**Did the same happen when all other 3rd party plugins are disabled?**
all my octoprints have the same software.

**Log file**
[octoprint (4).log](https://github.com/OllisGit/OctoPrint-FilamentManager/files/6215342/octoprint.4.log)
Drag and drop the octoprint.log here.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
Add any other context about the problem here.

jbbandos, 2021-04-02:
I can't connect to the database from filament manager/octoprint. It used to work, but now it fails the test, and the spools list is empty. From the command line, using "psql -h 127.0.0.1 -U octoprint octoprint_filamentmanager", the connection works well - but from octoprint I get this error in the logs:

> 2021-04-02 18:23:38,688 - octoprint.plugins.filamentmanager - WARNING - self.filamentManager is not initialized yet
2021-04-02 18:23:38,713 - octoprint - ERROR - Exception on /plugin/filamentmanager/database/test [POST]
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/flask/app.py", line 2447, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/pi/oprint/lib/python3.7/site-packages/flask/app.py", line 1953, in full_dispatch_request
    return self.finalize_request(rv)
  File "/home/pi/oprint/lib/python3.7/site-packages/flask/app.py", line 1968, in finalize_request
    response = self.make_response(rv)
  File "/home/pi/oprint/lib/python3.7/site-packages/flask/app.py", line 2098, in make_response
    "The view function did not return a valid response. The"
TypeError: The view function did not return a valid response. The function either returned None or ended without a return statement.
2021-04-02 18:23:38,734 - tornado.access - ERROR - 500 POST /plugin/filamentmanager/database/test (::ffff:10.0.0.143) 56.98ms

OllisGit, 2021-04-19:
Database driver is missing:
```
ERROR - Failed to initialize database: No module named 'psycopg2'
```

Look for ```psycopg2``` in this issue tracker (also closed issues)...sorry to not providing you a detailed answer...no time, need to go to bed ;-)

derekslenk, 2021-09-15:
@jbbandos did you have a chance to figure it out?

OllisGit, 2021-10-17:
fyi: the new version 1.9.0 shows more information, why the db-connection is not working.