OllisGit/OctoPrint-FilamentManager #55: Test connection to external databas posgresql not working

Doprintityourself, 2021-04-09:
I have setup everything like your guides tell.

I have 2 octoprint raspi 4 running in the same network.

connection is opended and set to my network range,
postgress database is running fine
but,
I cannot connect from both octoprint installations of filament manger to the database,
on the local install and on the remote install, the test connection newer turns green!

Any help appreciated
Thanks in advance

OllisGit, 2021-04-19:
@Doprintityourself 
please provide a octoprint.log. without more details I can't analyse the issue.

Xiondewah, 2021-04-30:
I've also got this issue - octoprint.log attached

Thanks.
[octoprint.log](https://github.com/OllisGit/OctoPrint-FilamentManager/files/6405646/octoprint.log)

Doprintityourself, 2021-04-30:
[octoprint (8).log](https://github.com/OllisGit/OctoPrint-FilamentManager/files/6406630/octoprint.8.log)
Tried with another instance to login for now even if the database is running and available on another device it either seem to not work at all and as an result of this is not working at all also no green database test button.

Doprintityourself, 2021-04-30:
ok another experience update:
 it did not work after hitting the test button (not green) and save!!
but 
after reboot the instance device is  now connected!
So would be interesting what the log above tells Olli.

Doprintityourself, 2021-04-30:
ashes on my head 👍 
may be I found an error why the filament-manger not worked after restarting octoprint service
I have forgotten to change the restart command to the seccond Instance (as I use up to 4 Instances per pie!)
will test!
but this does not change that green button and database does not work right after configuring the external database

Doprintityourself, 2021-04-30:
https://user-images.githubusercontent.com/70262325/116733053-e89dda80-a9eb-11eb-8aa0-d3bffcddd06d.mp4
so, final investigation: database is working if you restart the right service at once ;-) ,but 
it remains the button does not turn green on testing, to have a "at once" feedback if it is working

Hint, it was the first time that after hitting the button, that it did not show the text of the button.
 I think is a tribute of the screen capture tool.

So conclusion: The only problem still left :"button does not turn green"!

hope this helps

Xiondewah, 2021-05-01:
I'm afraid it's just not working for me.  Can't connect either on the pi with the DB or the second pi with updated connection settings.

OllisGit, 2021-06-05:
@Xiondewah, please take a look into your current octoprint.log.

Maybe there is still this issue:
```
octoprint.plugins.filamentmanager - ERROR - Failed to initialize database: No module named 'psycopg2'
ModuleNotFoundError: No module named 'psycopg2'
```
You have couple of options to fix this. Sometime a simple install helps (make sure you install it in the used Octoprint python):
```
pip install psycopg2
```

Sometime it is need to install also some dependencies
```
$ sudo apt-get update
$ sudo apt-get install python-dev libpq-dev
```
Or take a look into other issues with the same problem, like this: https://github.com/OllisGit/OctoPrint-FilamentManager/issues/10#issuecomment-706411613
Or https://github.com/OllisGit/OctoPrint-FilamentManager/issues/4#issuecomment-699649114

OllisGit, 2021-10-17:
fyi: the new version 1.9.0 shows more information, why the db-connection is not working.

OllisGit, 2021-11-15:
Hi @ZombiesLoveMe,

after the server starts the ```initialize``` is executed and throws this error: 
```
2021-11-09 14:03:55,343 - octoprint.plugins.filamentmanager - ERROR - Failed to initialize database: (psycopg2.OperationalError) could not connect to server: Connection refused
	Is the server running on host "192.168.0.34" and accepting
	TCP/IP connections on port 5432?
```
After that there is a follow up issue trigger from ```on_after_startup``` 
```
    all_selections = self.filamentManager.get_all_selections(self.client_id)
AttributeError: 'NoneType' object has no attribute 'get_all_selections'
```

So, please make sure your connection settings are correct.