OllisGit/OctoPrint-FilamentManager #56: Upgrade to support Python 3.4.9

jmlg10, 2021-04-12:
running Octoprint on a Windows PC running python 3.9.4, could it be possible to have an update of this plug-in?

Many Thanks

jmlg10, 2021-04-12:
[image]

OllisGit, 2021-04-13:
Hi @jmlg10,

please provide the full octoprint.log. I need to identify which function is not working. Maybe it has nothing to do with python 3.9.4.

jmlg10, 2021-04-14:
> WARNING: This is an external email. Shall you have any doubt, report any suspicious e-mail by clicking on the "Report suspicious e-mail" button in Outlook.

Hi Ollis,

Here is the Log

Let me know if you find something I can resolve.

Many Thanks

Cordialement / Best Regards

Jean-Michel LE GOFF | Responsable Recherches Numériques | Louis Vuitton Malletier | 2, Rue du Pont-Neuf, 75001 Paris, France | Phone: +33 155803048 | Mobile : +33 664405492 | Email address:  ***@***.***

jmlg10, 2021-04-14:
[octoprint.log](https://github.com/OllisGit/OctoPrint-FilamentManager/files/6309473/octoprint.log)
Hello,
Here is the log.
let me know if you find something I can resolve.
Many Thanks

OllisGit, 2021-04-19:
Hi @jmlg10,
found this in your logfile: 
```
AttributeError: module 'time' has no attribute 'clock'
```
duplicated issue, see #23 ...short python version downgrade....or wait for a fix

closed 2021-04-19 by OllisGit