OllisGit/OctoPrint-FilamentManager #57: Color selection for spools

ingo916, 2021-04-15:
**Is your feature request related to a problem? Please describe.**
No

**Describe the solution you'd like**
It would be nice to have a color picker and assign the corresponding color to the spool I have entered. also add a column header Color
**Describe alternatives you've considered**
None

**Additional context**

[image]

[image]

OllisGit, 2021-04-19:
This feature is implemented in SpoolManager and will not implemented in FM, sorry!

closed 2021-04-19 by OllisGit

ingo916, 2021-04-20:
SpoolManager currently does not support shared database. That’s why I switched to FilamentManager.