OllisGit/OctoPrint-FilamentManager #62: Plugin *seems* to ignore G2/G3 Moves

DJGummikuh, 2021-05-03:
**Describe the bug**
I am very new to python so I am not certain I read it right but while reading your code trying to learn for my own plugin I noticed that you are only appearing to treat "G0" and "G1" as move commands, completely ignoring Arc moves (which also extrude filament)

**Steps to reproduce**
1. Go to new_odometer.py
2. look closely
3. ...
4. Profit!

**Expected behavior**
Arc moves should be respected alongside linear moves

**Did the same happen when all other 3rd party plugins are disabled?**
yes, other plugins do not alter your python code :)

**Log file**
No

**Screenshots**
Screenshotting your source code appears dumb, so I don't :-)

**Additional context**
Thanks for allowing me to learn on your source ^^

OllisGit, 2021-06-05:
Will be in the next release, see https://github.com/OllisGit/OctoPrint-FilamentManager/pull/67

OllisGit, 2021-06-05:
Hi @DJGummikuh (geiler name)
PR is now in latest Version 1.8.1.

Please test and give feedback

DJGummikuh, 2021-12-05:
Damn sorry, for some reason I did not get any notification for that and only found out about your comment today >.<