OllisGit/OctoPrint-FilamentManager #63: cant install postgresql

Karico91, 2021-05-03:
hey there,
did anyone can help me?

pi@octopi:~ $ sudo apt install postgresql
[sudo] password for pi:
Reading package lists... Done
Building dependency tree
Reading state information... Done
The following additional packages will be installed:
  libllvm7 libpq5 postgresql-11 postgresql-client-11 postgresql-client-common
  postgresql-common sysstat
Suggested packages:
  postgresql-doc postgresql-doc-11 libjson-perl isag
The following NEW packages will be installed:
  libllvm7 libpq5 postgresql postgresql-11 postgresql-client-11
  postgresql-client-common postgresql-common sysstat
0 upgraded, 8 newly installed, 0 to remove and 28 not upgraded.
Need to get 14.7 MB/27.0 MB of archives.
After this operation, 105 MB of additional disk space will be used.
Do you want to continue? [Y/n] Y
Err:1 http://raspbian.raspberrypi.org/raspbian buster/main armhf libpq5 armhf 11.9-0+deb10u1
  404  Not Found [IP: 2a00:1098:0:80:1000:75:0:3 80]
Err:2 http://raspbian.raspberrypi.org/raspbian buster/main armhf postgresql-client-11 armhf 11.9-0+deb10u1
  404  Not Found [IP: 2a00:1098:0:80:1000:75:0:3 80]
Err:3 http://raspbian.raspberrypi.org/raspbian buster/main armhf postgresql-11 armhf 11.9-0+deb10u1
  404  Not Found [IP: 2a00:1098:0:80:1000:75:0:3 80]
E: Failed to fetch http://raspbian.raspberrypi.org/raspbian/pool/main/p/postgresql-11/libpq5_11.9-0+deb10u1_armhf.deb  404  Not Found [IP: 2a00:1098:0:80:1000:75:0:3 80]
E: Failed to fetch http://raspbian.raspberrypi.org/raspbian/pool/main/p/postgresql-11/postgresql-client-11_11.9-0+deb10u1_armhf.deb  404  Not Found [IP: 2a00:1098:0:80:1000:75:0:3 80]
E: Failed to fetch http://raspbian.raspberrypi.org/raspbian/pool/main/p/postgresql-11/postgresql-11_11.9-0+deb10u1_armhf.deb  404  Not Found [IP: 2a00:1098:0:80:1000:75:0:3 80]
E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?

thanks

-Karico

OllisGit, 2021-06-05:
Did you try:
```
apt-get update 
```
as suggested in the log-message? 

Or try this one:
```
sudo dpkg --configure -a
sudo apt-get upgrade --with-new-pkgs
```

OllisGit, 2021-10-17:
fyi: the new version 1.9.0 shows more information, why the db-connection is not working.