OllisGit/OctoPrint-FilamentManager #64: Confirm selected spool popup must be opened twice

StephanBeutel, 2021-05-03:
**Describe the bug**
I activated the "Confirm selected spool popup". On every print I have to open it two times as the first time there is no dropdown to select the spool.

**Steps to reproduce**
1. Activate "Show dialog to confirm selected spools before starting/resuming the print job"
2. Upload gcode file (file list must be empty before upload)
3. Click in "Start print"
4. Dialog does not show dropdown to select the spool

I delete the gcode files after printing. So mostly I start with an empty file list.
Same behavior is also if you do a server restart. On first print the popup has to be opened twice.

**Expected behavior**
Dialog shows the dropdown on first opening.

**Did the same happen when all other 3rd party plugins are disabled?**
Yes

**Log file**
Drag and drop the octoprint.log here.

**Screenshots**

OllisGit, 2021-06-05:
Hi @StephanBeutel,
is the spool-dropdown in the sidebar filled?
Please make a screenshot of the "missing selection-box".

Internal: There are a couplle of conditions for showing the selection-list:
```
!$root.core.bridge.allViewModels.printerStateViewModel.isPrinting() &&
$root.core.bridge.allViewModels.loginStateViewModel.isUser()
```

StephanBeutel, 2021-06-05:
Hi,
I already added screenshots, but for some reason they are missing.

The Dropdown in the sidebar is filled. I can also reproduce it by just disconnect and reconnect the printer.
Even when the list is filled with old prints. So the first start of a print job does not show the dropdown.

I know in the screenshot you can see, that I use some more plugins, but I already deactivated all plugins before I created this issue. Don't want to do this again, as the behavior is the same.

[image]

When I close this popup and just click on "start print" again, then the dropdown appears.

[image]