OllisGit/OctoPrint-FilamentManager #69: Latest update conflicting with other plugins and not working properly.

asylumlife, 2021-06-06:
I installed the new update.
Now when I boot up many of my plugins are not acting properly.
Themeify and others are not changing the look and several others are not showing up.

When I go into Filament Manager to check the settings and click on the setting button it does nothing. 

Also none of my filaments show up in the dropdown. I am using postures.

Xfirdan, 2021-06-08:
I have similar problems caused by Filament Manager after the update.
It conflicts somehow with touchUI so that local connections can´t be established correct.

asylumlife, 2021-06-15:
I have disabled every plugin I can and still have conflicts. I really like this plugin and would love to continue using it. Am I in the wrong place to find out about this issue?

asylumlife, 2021-06-29:
This is video of what is happening.

https://user-images.githubusercontent.com/85191866/123724174-e0321680-d859-11eb-8201-42351c362ca2.mov

asylumlife, 2021-07-21:
I have updated to the latest Pi OS and made sure all other items are up to date. I am still having the same issue.

asylumlife, 2021-07-21:
I do not have TouchUI installed.

kami83, 2021-07-21:
Same Problem here. TouchUI on my TFT is broken, when Filament Manager ist active.

asylumlife, 2021-08-21:
I have totally rebuilt my Octoprint for one of my printers. Now I can't even get it it to connect to the same postures it connected to earlier

asylumlife, 2021-08-29:
I started another install of octoprint on another raspberry pi. Now when I go to the plugins to add filament Manager it does not appear in the new plugins area.

nashuar, 2021-09-04:
I found a quick&dirty fix on my Octoprint instance :-)
I modified the file /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_filamentmanager/static/js/filamentmanager.bundled.js
on line 324 which contains
            var currentSelectionForTool = allCurrentSelections?.[toolID]?.name;
and replaced it with
            var currentSelectionForTool = allCurrentSelections[toolID].name;
and restarted Octoprint.

It's possible that this change breaks the spool confirmation dialog. Since I'm not using the confirmation dialog, I cannot confirm if it's broken or not. My local chromium instance on the Octoprint host now loads the site like it should. 

Maybe this helps you out :-)

OllisGit, 2021-10-17:
Hi @nashuar,
I think your mentioned changed is not a good approach, because this could raise a new error. The ? should prevent this. You already mentioned this.

Also this change has (should) nothing to do that the configuration-dialog is not opened (seen in Video).

So, I need more details.

Please take a look into the Browser Developer-Tools (open via Browser-Menu or pressing F12), maybe there is an error in the console.

@asylumlife what versions to you use in your new environment (op, python)...maybe FM is not working with the newest version.
Please take a look into the ```octoprint.log``` file

asylumlife, 2021-11-14:
@OllisGit Sorry never got an alert to this message. Not sure what happened. I will look at the version of the database I am running when I get home but filament manager was working with that database. 2 updates ago it broke and even with the latest update it does not work. I'll let you know.

asylumlife, 2021-11-16:
@OllisGit Postgress 12.

boinappi, 2021-11-24:
Experienced similar issues in combination with [Enclosure](https://plugins.octoprint.org/plugins/enclosure/)
Enclosure lost all preferences and ability to turn GPIOS on and off.

OllisGit, 2022-01-12:
@boinappi do you have an explanation about this "collision"? 

@boinappi , @asylumlife  please attach  the ```octoprint.log``` maybe I can find the error.