OllisGit/OctoPrint-FilamentManager #71: G Code Filament Selection

demitrix, 2021-06-20:
PrusaSlicer allows for custom start G-code per filament. Using this filament manager could automatically select the correct spool. This would be useful for print jobs started from outside OctoPrint when changing filaments.

ldiegos, 2021-07-24:
Hi,
I like this idea.

You could add some program lines to search some like:
**;FILAMENT_MANAGER: {filament_settings_id[0]}**

In my case, this line generate something like:
;FILAMENT_MANAGER: Warhorse-Black-PETG@Asun3Pro 

Where @Asun3Pro is the name of the printer. This the way i identify which filament belongs to each printer, but could be replaced with some other pattern that your could ignore and only take into account of "Warhorse-Black-PETG"

Thank you so much in advance.

OllisGit, 2021-10-17:
This feature is already tracked in the SpoolManager-Backlog (https://github.com/OllisGit/OctoPrint-SpoolManager/projects/1) ... unfortunately for you, this is tracked in the column "mid-term". So it will take a while to complete.

wernight, 2021-12-16:
I'd like to improve my change filament macro (using Klipper but that shouldn't be relevant) so that when I run it, it tells the Spool Manager to set that no filament has been selected.

Of course ideally I could also scan a QR or input some ID via MQTT to set the current spool, or scan via RFID... those are extensions especially if it can block the print until I've manually set a spool.

OllisGit, 2022-01-12:
fyi: @wernight "scanning" spools is also already tracked: https://github.com/OllisGit/OctoPrint-SpoolManager/projects/1#card-27695769

badnetmask, 2024-04-15:
So, I came here because I found issue#110, which is what I wanted to track. Looking at the GCODE generated by Prusa Slicer, I have the following lines.
```
;COLOR_CHANGE,T0,#E8AA2A
M600
```
So it seems like Prusa already injects the color I have selected in the UI, even though it doesn't necessarily matches the color code of the filament. Maybe that helps?

Also: I tried looking at the project list under the "mid-term" column and could not find this request. So where is it being tracked?