OllisGit/OctoPrint-FilamentManager #76: Multi Pi Filament Tracking

Ritchieeee, 2021-08-25:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

Ritchieeee, 2021-08-25:
Hey, Firstly, I love your work, your plugins have made my life so much easier. 

But would you consider making the spool tacking plugin capable of linking to multiple Rasp Pi's running octoprint?

I'm relatively new to octoprint, and on Github no more than 5 minutes, so apologies if you have this already answered somewhere (im still learning the navigation on this site!)

Kind Regards, 
Ritchie

derekslenk, 2021-09-15:
Some testing I did: I installed filament manager on one of my pis, and connected it to a database that was on a different server I have in my house. It created the four tables correctly and seems to be tracking changes correctly.

I installed the plugin on my other pi, pointed it at the same database with all the same information, and it will not connect. There are errors thrown (can attach if wanted)

So, while the database should allow for the potential, it does not currently seem functional.