OllisGit/OctoPrint-FilamentManager #85: Filament Manager Console Error

cgcarter1, 2021-10-30:
**Describe the bug**
Filament Manager has an error while validating the float.  There is an unrecognized token in the code.  It completely makes OctoPrint useless as the UI refuses to load.  The problem is in filamentmanager.bundled.js 

**Steps to reproduce**
1. Enable Filament Manager
2. See error

**Expected behavior**
The OctoPi UI to load

**Did the same happen when all other 3rd party plugins are disabled?**
No.  Only happens when Filament Manager is enabled

**Log file**
Drag and drop the octoprint.log here.

**Screenshots**
[image]
[image]

**Additional context**
Add any other context about the problem here.

OllisGit, 2021-11-14:
Hi @cgcarter1 ,

unfortunately in you screenshot the mentioned error-line 16959 is not visible, so I have no idea whats going on there.

biancapascal, 2021-11-17:
Here the same problem, Octoprint just starts in save mode till i disable Filament Manager.
Previous version had no problem.

OllisGit, 2021-11-20:
Hi @biancapascal.
please provide more details:
- octoprint.log
- filament database (if you use file based db), located: ```<>OP/data/filamentmanager/filament.db```

cgcarter1, 2021-11-20:
> Hi @cgcarter1 ,
> 
> unfortunately in you screenshot the mentioned error-line 16959 is not visible, so I have no idea whats going on there.

Sorry. The line number should be 16655. It is conflicting with other plugins with your validateFloat function.  Completely kills the UI. I disabled the plugin and the UI works again.