OllisGit/OctoPrint-FilamentManager #86: Not showing up anymore

michellauser, 2021-11-05:
I have the Problem, after update OctoPrint to 1.7.2 Filament Manager is not shwoing up ind OctoPtrint anymore - tired deactivate - activte, uninstall, reinstall - nothing works anymore....

OllisGit, 2021-11-14:
Please attach the octoprint.log file maybe there is an error message.

michellauser, 2021-11-15:
Hi Olli - thanks for your reply - found the error yetserday. While updating OctoPrint and invorenment to latest versions i change the update channel to Maintenance RCs an forgot to set it back - this was the rerason why the filament manger didn't work anymore. By changing the settings back to release channel stable, i was able to reinstall filament manger and import backup. Everything is working fine now.

Best regards Michael

closed 2021-11-15 by michellauser