OllisGit/OctoPrint-FilamentManager #98: Change filament after a print started

eadi, 2022-03-21:
I always forget to set the correct filament after I loaded a new spool. I have to restart the print or write down the current amount to fix it manually afterwards... This could be easily solved by code:

* During a print, the total filament usage of this print is not only saved in the database but also in a local variable ("locally stored filament usage variable")
* The filament selection dropdown is not disabled during a print
* When changing the filament during a print, a popup asks the user, if he wants to change the filament also for the already printed part
  * If the user chooses NO, the locally stored filament usage variable is reset and the rest of the print is substracted from the newly selected filament
  * If the user chooses YES, the locally stored filament usage variable is added to the previously selected filament again and substracted from the newly selected filament
* When the print is finished, the locally stored filament usage variable is cleared.

I think such a feature would create a much better user experience with (from my novice estimation) a relatively low implementation effort and I would personally be very happy with it.

Thank you for reading this!

bushibot, 2022-04-07:
Really need this, I'm swearing up a storm right since it will halt a print job because I forgot to update after swapping an almost empty roll.

eadi, 2022-04-29:
After realizing that this plugin is abandonned and a new one "Spool Manager" is around I tested it just to find out that this feature is already there!

closed 2022-04-29 by eadi