OllisGit/OctoPrint-FilamentManager #104: Can't confirm Spool selection if not set before

yani66, 2022-08-21:
**Description**
If you want to use the dialog to select the filament you are using before a print starts you need to select the spool you wan't to use before. If you have a different spool selectet you can't press start.

**Steps to reproduce**
1. Enable "Show dialog to confirm selected spools before starting/resuming a print job"
2. Select a filament in the sidebar
3. Start a print
4. In the dialog choose a different filament
5. Can't continue/press start

**Expected behavior**
I would expect it to ignore the previous selection from the sidebar and to press start with every spool you can choose.

**Screenshots**
[image]

hwmland, 2022-09-18:
But this is whole point of this confirmation I think.
The true is that it would be interesting option to allow making selection always only in this confirmation dialog.

yani66, 2022-09-18:
Yes, this would be good option, because if it's only a confirmation it wouldn't need a drop-down menu too select the active spool.