OllisGit/OctoPrint-FilamentManager #112: FIlament Manager across multiple instances of Octoprint

AndyShap2020, 2022-12-28:
I LOVE this plugin! I've just purchased not one, but two more printers. Each will have its own instance of Octoprint. But, I am now realizing I am going to have a problem:  If I use a spool of filament on one printer, and then transfer that spool over to another printer, the amount left is now screwed up. Is there any way to have the plugin talk to other instances of Octoprint and keep track of spool usage across printers?

ericus256, 2023-01-10:
@AndyShap2020 , this is a feature (see wiki instructions for [raspbian](https://github.com/OllisGit/OctoPrint-FilamentManager/wiki/Setup-PostgreSQL-on-Raspbian-(Stretch)) or [Arch Linux](https://github.com/OllisGit/OctoPrint-FilamentManager/wiki/Setup-PostgreSQL-on-Arch-Linux).)

However, currently this feature has a bug that renders it (somewhat) broken. See #106 & #112. You can share a common database so you don't have to enter in each spool by hand into each Octoprint instance, but (AFAIK) you can't "set" each printer to a different spool in Octoprint.... which, doesn't make much sense :)

AndyShap2020, 2023-01-10:
Hi Eric,  actually, I may have asked the question inappropriately.  I don't want to set each l to a printer. I want the central database.  So if I put spool #2 on printer #1, Octoprint will know how much filament is still on the spool.  Then I could move spool # 2 over to printer #1 and print a job on that printer.